opendata-stuttgart/sensors-software #620: firmware update NRZ-2020-129 (Jan 7 2020 13:43:50)  HTML read problem

Rewop-jpc, 2020-01-09:
After the firmware update NRZ-2020-129 (Jan 7 2020 13:43:50) I cannot read the http://192.168.1.125/values page  with VB.net to retrieve the values for my software.

dirkmueller, 2020-01-09:
for machine readable version, /data.json should be used instead. 

what is the issue when you try reading it? can you access it manually? is there an error message that would give us a hint on what's going on?

Rewop-jpc, 2020-01-09:
I use this with other html pages in VB.net. Until now it worked fine:

        Dim client As WebClient = New WebClient
        Dim data As Stream = client.OpenRead("http://192.168.1.125/values")
        Dim reader As StreamReader = New StreamReader(data)
        Dim str As String = ""
        Dim text As String = ""
        Do
            str = reader.ReadLine()
            text = text & str
        Loop Until str Is Nothing

After that I retrieve te values i want.

I have not used the json before?

Rewop-jpc, 2020-01-09:
The 'text' is fully filled, but you can see only the first part. 
The real reason was:
Number of measures is removed from the value page

Solved

dirkmueller, 2020-01-09:
Yes, this is now shown in /status instead. If you want I can move this value back to /values , I kinda didn't expect anyone already using it there because I just added it a few months ago. The idea was that /data.json and /values are now similar and sensor centric. Device centric data is now under /status.

dirkmueller, 2020-01-09:
let me know if there is anything remaining.

closed 2020-01-09 by dirkmueller

Rewop-jpc, 2020-01-09:
I see where it is now. I used it to see if the unit was recently reset/restarted.