opendata-stuttgart/sensors-software #621: firmware NRZ-2020-129 reads far to high PM values

tmeidinger, 2020-01-09:
NRZ-2020-129 reads PM10 and PM2.5 values >999. Reboot warm/cold didn't fix the issue. I installed NRZ-2020-130-B1 and readings are back to normal (<10).

dirkmueller, 2020-01-09:
@tmeidinger NRZ-2020-130-B1 and NRZ-2020-129 are other than the version number bitwise identical. 

can you please OTA downgrade again to 2020-129 and reproduce the 999 reading issue?

tmeidinger, 2020-01-09:
In the meantime NRZ-2020-130-B1 also produced far to high readings. Did the OTA downgrade to non-beta now.
[image]

tmeidinger, 2020-01-09:
{"software_version": "NRZ-2020-129", "age":"26", "sensordatavalues":[{"value_type":"SDS_P1","value":"600.43"},{"value_type":"SDS_P2","value":"421.27"},{"value_type":"temperature","value":"14.70"},{"value_type":"humidity","value":"45.00"},{"value_type":"samples","value":"4402602"},{"value_type":"min_micro","value":"32"},{"value_type":"max_micro","value":"20276"},{"value_type":"signal","value":"-59"}]}

dirkmueller, 2020-01-09:
@tmeidinger sorry for asking the obvious, but did you check that there isn't something physically blocking the air suction of the sensor or that the fan is still running every measurement interval? 

In my experience measurement values of 999 are an error code from the sensor. 

Please attach a screenshot of the /status page.

Then, could you please set debug to max and capture the serial output of the esp8266 for a few minutes while it is measuring and attach it here?

tmeidinger, 2020-01-09:
[image]

From what I can see the airflow is fine. For further inspection I'd need to get a ladder and dismount the sensor. Maybe something I'm going to do over the weekend.

tmeidinger, 2020-01-09:
After having power cycled again the SDS011 is reading ERROR now. Seems to be an hardware issue. Sorry for having you bothered.

closed 2020-01-09 by tmeidinger

dirkmueller, 2020-01-09:
What we can see in this screenshot is that the sensor was reporting this without CRC errors before. so 999 is some kind of error code (unfortunately it isn't documented what it is). as CRC was matching it is unlikely to be a issue with the firmware. 

I had this once myself already and there it was caused by an blocked suction pipe. Hope you find the reason.