opendata-stuttgart/sensors-software #626: Device is not working with hidden SSID on channel 11-13

DeeKey, 2020-01-11:
Device is not working with hidden SSID on channel 11-13
To reproduce
1) Hide SSID and switch it to channel 11-13
2) Load and configure old stable version (125 or earlier)
3) Initiate update to stable

Expected:
1) After update sensor will reconnect to hidden WIFI

Actual:
1) Sensor is not able to connect to hidden WIFI with channel from 11-13
2) If you make SSID visible or change channel it will connect successfully

DeeKey, 2020-01-11:
Seems that this is SDK problem according to @dirkmueller

dirkmueller, 2020-01-12:
Fixed in NRZ-2020-130-B2 or newer (just subscribe to the beta channel)

dirkmueller, 2020-01-12:
Reopen until landed in stable

closed 2020-10-17 by ricki-z

DeeKey, 2020-10-19:
Is it finally landed in stable?

ricki-z, 2020-10-19:
@DeeKey should be in stable now, but I haven't tested it yet.

commit a5c15584 (2020-01-12): IPv6 capable LWIP and Up to Wifi Channel 13 (Fixes #626)