opendata-stuttgart/sensors-software #628: Support for ZMOD4510?

rindlerblabla, 2020-01-12:
Just stumbled across this nitrogen oxide and ozone sensor, available at mouser for just about 10 euro. Don't know more about it than it's pretty new. Have you considered this sensor?

https://www.idt.com/us/en/products/sensor-products/gas-sensors/zmod4510-outdoor-air-quality-sensor-platform

DeeKey, 2020-01-12:
It outputs Mox resistance
https://www.manualslib.com/manual/1637921/Idt-Zmod4510.html?page=1#manual

dirkmueller, 2020-01-15:
@rindlerblabla to me this looks interesting, thanks for pointing this out!

rindlerblabla, 2020-01-20:
I got the rpi code from Renesas (there's no available arduino code yet). Maybe to be expected Q2. Don't have the technical skills to convert the code myself to be able to use on an arduino. The rpi code is for personal use, though, but the support guy at Renesas asked about getting in contact with you. Maybe the easiest way is to open a ticket at https://en.na4.teamsupport.com/dashboard

rindlerblabla, 2020-01-23:
And obviously the product from mouser that I was referring to is just the chip, not installed on a breakout board (https://eu.mouser.com/ProductDetail/Renesas-IDT/ZMOD4510AI1V?qs=sGAEpiMZZMtWSrBd5SaE4KIbT1yLGXlloupcWNifC7qBYa%2FyTHMPOQ%3D%3D). The image is just for reference, it's not the product they're selling. So wait until you find one with breakout board before ordering.

JanBosNL, 2020-01-29:
There has been some research about cheap NOx censors done by the (in dutch) dutch authority RIVM. https://www.samenmetenaanluchtkwaliteit.nl/sites/default/files/2017-09/RIVM%20notitie%20meten%20van%20NO2%20met%20goedkope%20sensoren.pdf

Problem they encountered witha lot of cheap NOx sensors is that they dont have enough low end sensitivity and a rather small resolution. To see if this zmod sensor is capable and accurate enough we would need to research by setting up a sensor side by side with a calibrated EU certified NOx sensor. 

Also NOx sensors can be very sensitive to moisture and other gasses in the air, and sometimes degrade rather quickly by use in the open air.

Some more information on these cheap NOx sensor tests (in dutch) here. 

https://www.samenmetenaanluchtkwaliteit.nl/sensoren-voor-no2

JanBosNL, 2020-02-13:
Anybody working on or testing this sensor?

rindlerblabla, 2020-02-13:
I am in contact with IDT and will probably start testing it soon, but I can only compare with PM2.5/10 and VOC. Don't have any possibility to compare it with any official sources.

JanBosNL, 2020-02-13:
Where do you live? 

You could for instance check if there is NOx data stations near your measurement location. Often this data is open and freely accesible so you can compare your data collected with the ZMOD4510 with it. 

For instance https://www.luchtmeetnet.nl/meetpunten?component=NO2&station=NL10742 is data collected by a station in 
the Netherlands. 
In the city of Nijmegen
At the street Ruyterstraat

rindlerblabla, 2020-02-13:
I live in Sweden, and my city's only got one official NO2 sensor and it is situated near one of busiest streets in the city (luckily don't live there), and they're not publishing any open data, just publishing a report once a year, so there are obviously some obstacles.

JanBosNL, 2020-03-08:
I found a first writeup of Pratik at PCBartists that tried to get the sensor working. http://pcbartists.com/design/sensors/getting-started-with-zmod4510/

DeeKey, 2020-07-15:
Any progress on the issue?

rindlerblabla, 2020-09-28:
I had quite a lot of contact with IDT and got some firmware. However I never got (or could manage myself) a working 8266-example. So I still haven't tried the sensor.

dslfkdslk, 2021-01-17:
There is the software for download 
https://www.file-upload.net/download-14443130/REN_Gas-Sensor-Evaluation-Software-7p0p1_SWR_20200608_1.zip.html

rindlerblabla, 2021-05-09:
Now there's an UART sensor available here, https://pcbartists.com/product/zmod4510-module/, which seems to be a lot easier to integrate.

closed 2026-09-26 by rindlerblabla