opendata-stuttgart/sensors-software #630: SPS30 connection issue

Petarkir2000, 2020-01-12:
Hello,
I try to connect the sensor SPS030, but I fail. 
I connect the sensor to it in the same way as SD011, but unfortunately no data.
Could you please tell me where I am wrong?
Attaching photos.

[image]
[image]
[image]
[image]
[image]

stewecar, 2020-01-12:
Hello,
the SPS30 configuration is not the same as SDS011..

**SPS30 (I2C, 5V)**

Pinout: 1 2 3 4 5

    Pin 1 (5V) -> Pin VU/VIN
    Pin 2 (SDA) -> Pin D3 (GPIO0)
    Pin 3 (SCL) -> Pin D4 (GPIO2)
    Pin 4 (SEL) -> Pin GND
    Pin 5 (GND) -> Pin GND

**SDS011 (serial)**

Note: Serial connections are always crossed (RX on one side is connected with TX on other side)

    Pin 1 (TX) -> (RX) Pin D1 (GPIO5)
    Pin 2 (RX) -> (TX) Pin D2 (GPIO4)
    Pin 3 (GND) -> GND
    Pin 4 (2.5m) -> unused
    Pin 5 (5V) -> VU
    Pin 6 (1m) -> unused

closed 2020-01-12 by Petarkir2000

svenkubiak, 2022-04-28:
@Petarkir2000 
How did you end up wiring the SPS30 **and** the BMP/BME280? Afaik, the BMP/BME280 is also connected to D3/D4:
[image]

pjgueno, 2022-04-28:
It is I2C. Same pins for both the buses.