opendata-stuttgart/sensors-software #632: Question - Android Mobile Feinstaub Source code?

CameronLamont, 2020-01-14:
[image]

Hello I have been searching for the android source code to the app in the photo.  It directed me to this repo but the mobilefeinstaub directory doesn't appear to contain an android project.

The apk doesn't appear to read from my sds011 and wanted to try and get it working.

ricki-z, 2020-01-15:
The "app" was developed with cordova (mentioned in the readme.md). Its some kind of webapp with access to hardware resources.
So it's not a native android app.

CameronLamont, 2020-01-15:
OK thankyou, I couldn’t see mention of this app in the readme. I will look into Cordova to see if I can make use of that instead.

From: Rajko Zschiegner
Sent: Wednesday, 15 January 2020 7:05 PM
To: opendata-stuttgart/sensors-software
Cc: CameronLamont; Author
Subject: Re: [opendata-stuttgart/sensors-software] Question - Android MobileFeinstaub Source code? (#632)

The "app" was developed with cordova (mentioned in the readme.md). Its some kind of webapp with access to hardware resources.
So it's not a native android app.

ricki-z, 2020-01-15:
Cordova is not an app. It's an open-source mobile development framework.

CameronLamont, 2020-01-15:
Understood, just haven’t used that framework before.  Thanks.

From: Rajko Zschiegner
Sent: Wednesday, 15 January 2020 7:20 PM
To: opendata-stuttgart/sensors-software
Cc: CameronLamont; Author
Subject: Re: [opendata-stuttgart/sensors-software] Question - Android MobileFeinstaub Source code? (#632)

Cordova is not an app. It's an open-source mobile development framework.

dirkmueller, 2020-01-15:
@CameronLamont any question remaining?

dirkmueller, 2020-01-17:
Please don't hesitate to reopen when there is something left..

closed 2020-01-17 by dirkmueller