opendata-stuttgart/sensors-software #635: Is it possible to push data to a running Domoticz instance?

mmartinortiz, 2020-01-16:
Hi,

I recently assembled my luftdaten.info device, and I was wondering if there is an easy way for connecting with my Domoticz server.

ricki-z, 2020-01-16:
Look at:
https://www.domoticz.com/forum/viewtopic.php?t=23406

(First result of a google search for "luftdaten.info domoticz" ,https://www.google.com/search?client=firefox-b-d&q=luftdaten.info+domoticz)

mmartinortiz, 2020-01-17:
Thanks @ricki-z , good to point to Google, I'll take a look to the thread.

On [DuckDuckGo](https://duckduckgo.com/?q=luftdaten.info+domoticz) (just another web search engine) the link to the domoticz web page is not present, interesting, because so far I never noticed bad results using DuckDuckGo.