opendata-stuttgart/sensors-software #638: Sds011 will not write

wtrnl, 2020-01-18:
The sds011 will not write the data.
Sometimes the data will displayed and some times not.
[image]
See also the graph.
[image]

wtrnl, 2020-01-18:
I read the data every 5 minutes.
5 minutes data then 5 minutes no data

dirkmueller, 2020-01-19:
@wtrnl please provide a screenshot of the "Device Status" (/status) page after a bit of uptime (20-50 measurements min) and if possible the serial output (with max info level set).

wtrnl, 2020-01-19:
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/opendata-stuttgart/sensors-software/issues/638?email_source=notifications\u0026email_token=AOKCPJYTSTGYJJG5O6UNA33Q6QPHRA5CNFSM4KIT3QQKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEJKNX3I#issuecomment-575986669",
> "url": "https://github.com/opendata-stuttgart/sensors-software/issues/638?email_source=notifications\u0026email_token=AOKCPJYTSTGYJJG5O6UNA33Q6QPHRA5CNFSM4KIT3QQKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEJKNX3I#issuecomment-575986669",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> }
> ]

Hereby the printscreen.Serial output not possible.Verzonden vanaf mijn Samsung Galaxy-smartphone.

dirkmueller, 2020-01-19:
@wtrnl unfortunately the mail attachment didn't come through.. please try attaching it on the webui of github (copy&paste in the browser)

wtrnl, 2020-01-19:
[image]

closed 2020-01-19 by wtrnl

dirkmueller, 2020-01-19:
@wtrnl did you intentionally close this?

What can be seen is that one out of 131 measurement cycles failed to get data from sds011. 12 out of 131 failed to be able to send the data, the last time due to connection loss.

So this looks primarily like an issue with connectivity to the configured APIs.