opendata-stuttgart/sensors-software #639: influxdb data logging

FollowTheWizard, 2020-01-18:
1. I do not have a SDS011 installed on my 2nd ESP8266. 
It is also not activated in the configuration page.
However I see the values "samples", "max_micro" & "min_mirco" in my influx database.
[image]

How can I get rid of those values in the database?

2. Measurement names within influx database:
I think it would be great if "humidity" would be called "DHT22_hummindity" and "temperature" -> "DHT22_temperature".
This would make the naming more consistent.
Same goes for "BMP_pressure" -> "BMP180_pressure" (Found it in airrohr-firmware.ino)
Or would this break something?

dirkmueller, 2020-01-19:
1. The values that you're mentioning: samples and max/min micro are unrelated to SDS011. *samples* refers to the number of loop() executions between two measurements. There should not be large variations basically unless you change the measurement interval. The max/min micro values are the maximum and minimum *microseconds* between two consecutive loop() invocations. loop() is the busy loop of Arduino (the microcontroller that you're using) As such those measurements are entirely unrelated to SDS011, which is about particulate matter air sensoring. 

I don't really know why we're reporting these values to the APIs. It was added in https://github.com/opendata-stuttgart/sensors-software/commit/445ae7c75c44d24f75193386eac0a5010bb8cce5 but there was no rationale given.

2. I agree, the naming of the variables is suboptimal. this has annoyed me quite a bit as well in my setup. I just don't know of a way to fix this without breaking all existing users and API integrations :(

dirkmueller, 2020-01-19:
@ricki-z can we move these samplesmin/max to the /status page and remove it from the API reporting? or are those values used anywhere?

ricki-z, 2020-01-19:
These values are needed for sensors that use PWM modulation to write value (i.e. PPD42NS) or that need a minimum time between readings (i.e. GPS, otherwise we can't read it fast enough).
Without these values transmitted to us a debugging may be much more difficult. And not all users with a problem have direct access to the sensor.

dirkmueller, 2020-01-20:
@ricki-z where can I see those values? they're not reported to sensorcommunity, only to the optional apis. so madavi.de holds statistics about them?

ricki-z, 2020-01-20:
These values are saved at Madavi.de in CSV files (together with the transmitted measurements).
http://www.madavi.de/sensor/csvfiles.php