opendata-stuttgart/sensors-software #640: NRZ-2020-130-B2 - Bulgarian translation issues

Petarkir2000, 2020-01-19:
Hello,
Could you correct the translation in Bulgarian version of NRZ-2020-130-B2 as follow:

1. Window "Device status"
- Free Memory	=> Свободна памет
- Heap Fragmentation => Heap фрагментиране
- Last OTA 	=> Последен OTA
- Uptime	=> Непрекъсната работа
- Reset Reason 	=> Причина за Reset
- OTA Return	=> Отговор OTA
- Data Send	=> Изпратени данни
2. Window "Configuration", 
2.1 sub-window "Sensors"
More Сензори	=> Още сензори
2.2 sub-window "APIs"
Measurement	=> Измерване

Thank you.

dirkmueller, 2020-01-19:
Thanks, you highlighted a couple of strings that are currently not being i18n'ed correctly.

Petarkir2000, 2020-01-19:
If you are talking about the Heap and OTA, I think in Bulgaria people use both words, because of long explanation on our language, especially Heap ;).

dirkmueller, 2020-01-19:
No, I just said the strings are not translatable yet. For "Device Status" page strings I'm still unsure which of those will remain (e.g. whether Memory fragmentation remains), so I didn't want to go through the typing exercise to add the 16 translations for something that might disappear or get reworded.

closed 2020-02-06 by Petarkir2000