opendata-stuttgart/sensors-software #643: Probleme mit Messinterval >= 400

Sansibari, 2020-01-20:
Mit langen Messinterval fehlen zeitweise die Anzeige der Sensorwerte unter meine.lufdaten.info/sensors/....

Kurz nach neuem Messinterval sind sie wieder da, verschwinden dann wieder.
Das gleiche für die Anzeige auf der Karte.

Mit Messinterval ~ 600 erscheinen die Messwerte überhaupt nicht mehr, obwohl diese übertragen wurden.

Ich möchte gerne die Messintervalle auf 1000 setzen.

ricki-z, 2020-01-20:
Wir lesen die Daten aus einem aktuell 19MByte großen JSON File. Dies enthält die Daten der letzten 5 Minuten. Wenn wir in dieser Datei die Daten für die letzten 1000 Sekunden (rund 15 Minuten) speichern würden, wäre diese Datei aktuell schon fast 60 MByte groß.
Und das müsste auf dem Server bei jedem Aufruf der Sensor-Daten durchsucht werden.
Das ist uns leider mit der aktuellen "Hardware" nicht möglich zu leisten.

dirkmueller, 2020-01-20:
@ricki-z does it have to be json? 12000 sensors with 2 inputs times ~ 20 bytes field values per measurement should be less than 5MB for 15 minutes... (e.g. influxdb or simply an in memory key/value store (e.g. memcached))

ricki-z, 2020-01-20:
JSON can be handled with most programming languages.

And what if the next user is asking the same question for an interval of 30 minutes? Same discussion again? We use an interval of 2.5 minutes, the data.json contains the data of the last 5 minutes (for the case of a missing or failed transmission). If we set these times higher someone will ask for just larger intervals.

InfluxDB: was tested. crashed regularly. is optimized for many writes, less reads. Used for the graphics on the map where we need the data for a single ID.

Please remember that this is a project made by volunteers. So we have to keep most things as easy as possible. There was more than one case where we had to rewrite code because the original developer was using a not so common or frequently changing framework.