opendata-stuttgart/sensors-software #645: Relation between ChipID and sensorID (Location ID)

djim1, 2020-01-20:
how can I find the chipID in the json (https://data.sensor.community/static/v1/data.json). Ik can find the location/Sensor ID that is showing on maps.luftdaten.info
For a local luftdaten project we only have the sensroID's of the participants. I have a csv with all the Chip-ID's and I want to check if all sensors are active/registered/on the map. I don't heve the map ID of the sensor (startin with # on the map).

at this moment I only can use the date/csv files from madavi because I only have the chipID's. But I wont to use the luftdaten json/csv. But then I have to know the sensorID's that match the chipID's.

Now i'm loking for a way to link the ChipID (ESP8266-xxxxxxx) to de sensorID (#xxxxx).

can someone tell me how?

ricki-z, 2020-01-20:
There is no accesible way to link the chipID to the ID on the map.
With a known chip-ID and a relation to the ID on the map it would be possible to look for sensors (and with that for the owner). The chipID is part of the MAC address of the wifi interface of the ESP8266.

dirkmueller, 2020-01-26:
I think question had been answered? Don't hesitate to reopen when there's still something open.

closed 2020-01-26 by dirkmueller