opendata-stuttgart/sensors-software #646: Error displaying temperature

Phaet, 2020-01-22:
I am running the sensors since a few days now and everything is working quite well. 
But I noticed now that as soon as **temperature goes bellow zero** the panel shows **constantly a value of -3273,0 degrees** which is also ignored in the map.

Phaet, 2020-01-23:
The wrong value is not constant, but varying somewhat around 327n,n.
Reading the debug info on USB serial port shows just the same values, but no additional information like raw values from the DHT22 sensor.
I also inspected the code a bit. The values are read out in DHT.cpp, where also the float value is computed from the raw bytes (data[2] and data[3]).
I wonder whether the sensor might be defect?

Phaet, 2020-01-24:
Thinking twice it can not be the sensor, since two bytes are max 256. So the wrong values have to be the result of a wrong calculation or a memory effect.
But I cannot see anything suspicious in the code.

ricki-z, 2020-01-24:
Two byte unsigned are 65536 possible values (signed: -32768 to 32767), 1 byte are 256.

RikDrabs, 2020-01-24:
Two bytes signed interpreted as two bytes unsigned gives this result.

Phaet, 2020-01-24:
Oups, cringy.
But anyway, seems my posts are actually read.

Any ideas?

ricki-z, 2020-01-24:
At the moment there are more than 6000 DHT22 active in the network. And most of them show values below zero correctly. Actually you can see this on the map in northern Sweden and Finland.
So it seems more like a problem with this DHT22 than the firmware.

Phaet, 2020-01-24:
I am quite well aware of this fact.
I just wonder whether a defect sensor could produce a value _systematically_ wrong: when the actual temperatur is -t the displayed value is obviously -3276,8+t

ricki-z, 2020-01-24:
@Phaet could you please post a picture of your DHT22 here? As mentioned most DHT22 doesn't have this problem.

Phaet, 2020-01-24:
[image]
This is the DHT 22 module which I purchased at Segor together with the other parts (actually the picture from the store, not the one I have in use).
I already was wondering what might be the difference between the bare DHT 22 and the modul other than the latter can also be operated with 5V?

Phaet, 2020-01-25:
I have just ordered a replacement.

ricki-z, 2020-01-25:
Only to answer to the image comment: This is the right module. The small PCB contains a resistor between VCC and the data pin as a pull up. As the firmware is activating an internal pull up the DHT 22 could also be connected without that PCB.

dirkmueller, 2020-01-26:
We can add a debug output but it sounds like the value that is being read is 0x8000 or on that range. We're verifying the checksum so theoretically there shouldn't be a problem.

How long is the cabling between sensor and esp microcontroller? It should not be much longer than 10cm, as onewire is quite sensitive to interferences. It is quite unlikely to have distortion with matching checksum however..

Phaet, 2020-01-26:
Sorry for the delayed answer. I am off for a week and back Feb 1. Th sensor is off during this week. When back I will replace the DHT 22 with another part which hopefully will be arrived till then.
I was told that the benefit of the modul is the pin distance of 2.54mm which fits into break boards and are more robust compared to the sensor pins with a distance of 2mm.
The cables are less then 10cm, so I do not think this might cause the troubles.
And yes, since the logic of the DHT.cpp code delivers valid values with respect to the checksum in code[4] the sensor must produce values which are consistently wrong. Later in the code the values are just used and no calculations or bit manipulations take place.
I will keep you posted.
And yes: Thanks for looking into it.

closed 2020-01-27 by Phaet

Phaet, 2020-02-05:
Just for the records:
Back home I installed the replacement DHT 22. And surprise - it works!
In fact, it was the defect sensor producing these odd values.