opendata-stuttgart/sensors-software #648: Differences between mesured data and displayed data on madavi.de

ansim58, 2020-01-24:
Hi, 
the question is, wether the measured data from the sds011 are modified when presented in the madavi-diagrams.
I have tried to build up some diagrams from my own data and compared them with the diagrams downloaded from madavi. There are significant differences!
This is my setup: Sensor #26205. The data is sent to Luftdaten.info and to a mysql-DB on my own server.  It is saved in the DB without any modification.
The diagram is build with a php-program using jdGraph. Again the data is presented without any modification. Now the data from my DB has much higher values then the data presented by madavi. 
Please pay attention to the slightly shifted x-axis. See the differences in the pictures:
[image]

ricki-z, 2020-01-24:
At first you should check the "order" of the data. PM2.5 is part of PM10, so the values for PM2.5 can't be higher than those for PM10 by definition.
If you switch the graphs the difference isn't that high:
max. "PM2.5" in you graph around 80µg, max. PM10 in the madavi.de graph is around 80µg.
The graphs at madavi.de are created with RRDTools. And these tools show averaged values per "row". With 500 pixels width a pixel row fits 2.88 minutes. So in some cases the average of 2 measurement is shown. And this is normally lower than the maximum of these two values.

ansim58, 2020-01-24:
Okay! Thank you for the quick and detailed response!
Indeed it's my fault. Meanwhile i found this:
https://github.com/opendata-stuttgart/meta/wiki/Beschreibung-Datenformat
--> SDS_P1: PM10 / SDS_P2: PM2.5
So thank you very much and sorry for the inconvenience!
Andreas

closed 2020-01-24 by ansim58