opendata-stuttgart/sensors-software #653: "NRZ-2020-129" - compile error

FrankUlbrich, 2020-01-27:
Hello I tried to compile the airrohr firmware "NRZ-2020-129" 
from the master branch for NodeMCU 1.0 and I am getting the following error.

'Session' in namespace 'BearSSL' does not name a type
-----------------------------------------------------------------

In file included from C:\Users\HP\Dropbox\Frank\Downloads\arduino\projekte\WEMO_ESP\airrohr-firmware_2020\airrohr-firmware_2020.ino:156:0:

ext_def.h:48: error: 'Session' in namespace 'BearSSL' does not name a type

     BearSSL::Session* session;

Is there something what I could do to get rid of this compile error ?

Viele Gruesse
Frank

ricki-z, 2020-01-28:
Could you please check the installed Arduino Core? This should be the version mentioned in the Readme.md.

FrankUlbrich, 2020-01-30:
[image]

Hi,
I am compiling wir ARDUINO IDE 1.8.3  and selected "NodeMCU 1.0" as target board. Not sure what happens. I tried the same with "Generic ESP3266" with the same result. In the past I  did not have problems with older versions.

[image]

I will try it with a new GITHUB sanpshot.

FrankUlbrich, 2020-01-30:
I tried it wit a fresh downloaded  master snapshot from today. 
Can anyone confirm or test if it really compiles with  NodeMCE 1.0  (ESP3266), please ?

ricki-z, 2020-01-30:
What "Arduino Core for ESP8266" have you installed? This should work with 2.6.3, the latest stable release.

FrankUlbrich, 2020-01-31:
ricki, thanks a lot for your hint. I didn't understand the meaning Arduino Core.
The compiler works now with the latest Arduino Core:
[image]

This solved my probems, thanks a lot.

FrankUlbrich, 2020-01-31:
Upgraded my IDE to esp3266 vore version  2.6.3

closed 2020-01-31 by FrankUlbrich