opendata-stuttgart/sensors-software #656: Blueberrye BME280 sensors pin layout

jpkrooney, 2020-01-29:
Hi,
I bought a few Blueberrye BME280 sensors after seeing them suggested here: https://github.com/opendata-stuttgart/sensors-software/issues/137#issuecomment-331731434

However they arrived with no documentation and I have no idea of the pin layout - does anyone have this documented somewhere ?

ricki-z, 2020-01-29:
You can find the needed informations in the FAQ at luftdaten.info ("Connect other sensors").

jpkrooney, 2020-01-29:
No I could not find the answer there. I mean the pin layout of the cables from the sensor which are customised for this product.

After much, MUCH searching I found what I needed in a customer review of the Blueberrye BME280 on Amazon.de. Unfortunately I could not make a direct link to the photo but here is a screen shot for future reference

[image]

closed 2020-01-29 by jpkrooney