opendata-stuttgart/sensors-software #657: Setting a static IP address

Petarkir2000, 2020-01-30:
The station works normally, but for a number of reasons it cannot connect to a DHCP server and a static IP address must be specified.
However, for a number of reasons, this is not appropriate for the particular case (the network has a functioning DHCP server from another subnet and the station always connects to it) and  have to be set it manually.
Is there any way to do this through the web interface or with "handy" tools?

Marwe, 2020-03-30:
So you cannot configure the DHCP appropriately? This would be the natural thing to do, at least since DHCP seems to be implemented for your (sub)net.

Petarkir2000, 2020-03-30:
Hello Marwe, I do not want to use DHCP on my router -read my above remark. The question is could we set Static IP address on station.

ricki-z, 2020-04-01:
At the moment a static IP s not configurable.

urs8000, 2022-07-15:
when will it be possible again to set a static ip in the web-configuration?
within several applications it could run in a problem having the IP set via DHCP

urs8000, 2023-11-28:
let's bring this topic to the top again

Phaze-III, 2023-12-02:
The current published beta firmware can be configured with a static IP-address.