opendata-stuttgart/sensors-software #658: Thingspeak Interface

plastikjoe, 2020-01-30:
Hello, 
please add a possibility to send live the data from the sensor (esp & DHT 22 & SDS011) to the platform thingspeak. actually is possible to send data to Feinstaub-App, aircms.online, OpenSenseMap.
Firmware: NRZ-2020-129 (Jan 7 2020 13:40:51)

many Thanks

ricki-z, 2020-01-30:
Thingspeak would need MQTT support. Look for the issue comments regarding MQTT why we don't support this.

dirkmueller, 2020-02-03:
I have not had any chance in signing up for an account at thingspeak and any attempt at reaching their support failed so far. does the site still allow for new subscriptions?

zoomx, 2020-06-26:
> 
> 
> Thingspeak would need MQTT support. Look for the issue comments regarding MQTT why we don't support this.

There is a Thingspeak library and it works without MQTT, they uses REST API also.
[https://www.mathworks.com/help/thingspeak/rest-api.html](url)