opendata-stuttgart/sensors-software #659: Reboot loop doing OTAUpdate, after configuration.

vilhelmp, 2020-01-31:
I have followed the instructions to build a luftdata sensor. I am using a Wemos D1 mini (with CH340) v 3.1.0. 

- I installed the latest (latest_se.bin) by flashing with the esptool.py  `esptool.py --port /dev/ttyUSB0 write_flash -fs 4MB -fm dout 0x0 ~/Downloads/latest_se.bin` after this, I reboot/reset the device (with all sensors plugged in). 
- I can access the configuration and set up wifi etc, and it reboots again.
- Then I can again access the main landing page with config/home/debug level etc.
- However it keeps rebooting. So I set up the serial monitor in the Arduino SDK and get the following log showing it rebooting over and over. 
[log_2020-01-31-10:15.txt](https://github.com/opendata-stuttgart/sensors-software/files/4138461/log_2020-01-31-10.15.txt)
.
Note though that it does send data sometimes [https://www.madavi.de/sensor/signal.php?sensor=esp8266-4851274](https://www.madavi.de/sensor/signal.php?sensor=esp8266-4851274)

What can I do to figure out what this is?

jgysenbergs, 2020-01-31:
I seem to remember the same issue and fixing it by erasing the Wemos D1 mini to factory settings [1] or esptool.py --port /dev/ttyUSB0 erase_flash

Hope this helps.

[1] https://wiki.wemos.cc/tutorials:get_started:revert_to_at_firmware

dirkmueller, 2020-02-03:
I will try to reproduce this when I'm back. what I'm seeing is that you're setting an unusual flash mode (dout). we expect dio in the firmware we are building and accept also qio. I was assuming the arduino core is compensating for an incorrect flash mode in the sketch calculation but it seems it doesn't for the (very unusual) dout setting. 

to recover from your problem I suggest to reflash with -fm dio or -fm qio. *don't* run erase_flash beforehand otherwise you're losing all settings. just flashing over the previous firmware is good.

dirkmueller, 2020-02-03:
Please let me know if flashing with the correct mode resolves this issue.

vilhelmp, 2020-02-05:
Finally got around to try it out (started new job on monday), and it works! Thank you for the insightful help. Just to be clear for anyone else, I reflashed with 
`esptool.py --port /dev/ttyUSB0 write_flash -fs 4MB -fm dio 0x0 /path/to/latest_se.bin`
and that did it.

closed 2020-02-05 by vilhelmp