opendata-stuttgart/sensors-software #661: SPS30

pritt654321, 2020-02-03:
Seit 02.02. bringt der SPS30 keine Werte mehr. Lief vorher ca. eine Woche stabil mit der Betasoftware auf dem NodeMCU. Seit 31.01 gibt es offensichtlich eine veränderte Beta, kann der der Grund sein?

dirkmueller, 2020-02-03:
bitte poste doch ein screenshot der "http://airrohr-XXXXXX.local/status" website.

dirkmueller, 2020-02-03:
also, does it recover after a power disconnect for 30s and powering it on?

pritt654321, 2020-02-04:
hier einige Screenshots
[image]
[image]
[image]

pritt654321, 2020-02-04:
[image]

dirkmueller, 2020-02-05:
Thanks. So the 'check wiring' means the sensor isn't even responding to the initialization..

pritt654321, 2020-02-05:
Thanks Dirk, you were right,one of the resistors wasn‘t soldered proberly and i will reinstall everything on the weekend

closed 2020-02-06 by pritt654321