opendata-stuttgart/sensors-software #664: New feature for the pmapp

marcauberer, 2020-02-06:
Hey,
we're planning a new feature for the pmapp, where we want to scan the local network for possible active ESPs. For that, if would be good to scrape the configuration site to get the sensors chip id, its AP name and the checkbox flag, if the transmission to our servers are enabled.

So here's my question: Are we permitted to develop this feature and release it to production?
If you wish, I'll send you screenshots and the GitHub link to the exact position in the source code (as soon as it's implemented), where the app scrapes the config site.

There shouldn't be security issues, cause e.g. the wifi pw doesn't get displayed in the html code of the page.

Looking forward to your reply!

ricki-z, 2020-02-07:
Shouldn't be a problem as long as the user is "asked". But why do you need the AP name?

marcauberer, 2020-02-07:
Thanks for your anwer. We want to display a identification name for each found device. So, the AP name seems fine for that, especially cause it's customizable for the user. Many users can not identify their sensors by the chip id, the ip and the mac address.
Thanks for your permission! so we start with the development.

closed 2020-02-07 by marcauberer

ricki-z, 2020-02-07:
I don't know it the AP name is really helpful. Most FritzBoxes and Telekom routers use the same name (ending with the model number), some routers use cryptic names (i.e. WL465288) that nobody can remember ...

marcauberer, 2020-02-07:
I think "AP-name" was the wrong term. I ment the name of the AP in the configuration mode:
[image]

marcauberer, 2020-02-12:
If someone's interested, we just released the [new update](https://chillibits.com/redirect?l=pmapp-play&ref=github) with the scanning feature included.

More info:
https://github.com/ChilliBits/particulate-matter-app
https://chillibits.com/pmapp