opendata-stuttgart/sensors-software #665: Big difference between DHT22 and BMP/E280

jsturm, 2020-02-07:
I have 2 temp sensors next to each other installed in my airrohr: DHT22 and BMP/E280
The readings are verry different. 

DHT22 | Temperatur | 8.8 °C
DHT22 | rel.   Luftfeuchte | 88.4 %
BMP/E280 | Temperatur | 11.1 °C
BMP/E280 | rel.   Luftfeuchte | 60.9 %

Is tis normal? 
The DHT22 seams to show the more realistic values, compared with the values of the local meteorological station.

Achim

zoomx, 2020-06-26:
BMP280 and BME280 are known to overestimate temperature so datasheet and Bosh site describe them as Perssure sensor and Pressure and Humidity sensor.
BME280 has better perfomance on Humidity than DHT22
[http://www.kandrsmith.org/RJS/Misc/Hygrometers/calib_many.html](url)
[https://wiki.liutyi.info/display/ARDUINO/v9+Sensors+Board](url)

ricki-z, 2020-07-03:
The overestimation for the BMp/E280 is only in "normal" (continuous) measurement mode. In continuous mode the Bosch sensors will become warm and the measured temperature is getting higher.
This can be disabled and we do this in our firmware. Take a look at this device: https://api-rrd.madavi.de/grafana/d/q87EBfWGk/temperature-humidity-pressure?orgId=1&var-chipID=esp8266-1027427
The temperature of the BME280 is lower than those of the DHT22.

stickybyte, 2020-09-26:
@ricki-z how is that working in the firmware? I dont see a switch for that.

zoomx, 2020-09-27:
This instruction
bmx280.takeForcedMeasurement(); (about line 2698 in main sketch)
mean that measure is done only when requested.
See datasheet.