opendata-stuttgart/sensors-software #668: PM values not correct

cristianbriscu, 2020-02-13:
The PM values are 10 times higher. Used the serial to usb adapter that came with SDS011 and measured about 6 ug/m3 but the software reported values like 60 ug/m3. Also the room purifier reads about 5-6 ug/m3. software version NRZ-2020-129. 

PM10 (sec.) : 57.10
PM2.5 (sec.): 53.00
PM10:  56.75
PM2.5: 52.60
----
End reading SDS011
## Sending to sensor.community - SDS011
Succeeded - api.sensor.community
Start reading DHT22
Temperature (°C): 25.00
Humidity (%): 28.70
----
End reading DHT22
## Sending to sensor.community - DHT22
Succeeded - api.sensor.community
## Sending to madavi.de:
Succeeded - api-rrd.madavi.de
## Sending to opensensemap:
Succeeded - ingress.opensensemap.org
## Sending as csv:
CSV Output: {"software_version": "NRZ-2020-129", "sensordatavalues":[{"value_type":"SDS_P1","value":"56.75"},{"value_type":"SDS_P2","value":"52.60"},{"value_type":"temperature","value":"25.00"},{"value_type":"humidity","value":"28.70"},{"value_type":"samples","value":"4388685"},{"value_type":"min_micro","value":"32"},{"value_type":"max_micro","value":"40750"},{"value_type":"signal","value":"-27"}]}
2225332;56.75;52.60;25.00;28.70;4388685;32;40750;-27
Time for Sending (ms): 3365

Parameter | Value
-- | --
Firmware version | NRZ-2020-129ST:1SDK:2.2.2-dev(38a443e)Core:2.6.2=20602000lwIP:STABLE-2_1_2_RELEASEglue:1.2-16-ge23a07eBearSSL:89454af
Free Memory | 26792
Heap Fragmentation | 5 %
Last OTA | 38 min, 3s
NTP Sync | 1
NTP Info | 91.216.151.61 (0.pool.ntp.org) reachable: Yes
Time | Thu Feb 13 11:27:01 2020
Uptime | 38 min, 4s
Reset Reason | Software/System restart
 
SDS011 | 18-11-16(25b6)
 
only errors
WiFi | 0
OTA Return | 200
Data Send | 0
SDS011 | 0
 
Number of measurements | 15
Time spent uploading | 3365 ms

ricki-z, 2020-02-13:
Please check this with another USB wall charger / power supply. Some of them limit the current to 100mA. In this case the laser is slightly darker and the measured values are too high.

cristianbriscu, 2020-02-13:
thank you for the quick reply, and my bad for not double checking stuff correctly. was not connected to the 5V but VIN aprox 4V. thank you for the info.

closed 2020-02-13 by cristianbriscu