opendata-stuttgart/sensors-software #669: BME280 not showing Temperature/Pressure/Humidity data ONLY on maps.sensor.community

Spaarne, 2020-02-14:
I replaced the SDS011 in my Luftdaten dust sensor with a BME280. The BME280 shows the measured Temperature, rel. Humidity and Pressure data fine on the sensor webpage. Also these data ar shown on Opensensemap.

However the T/H/P data are NOT shown on the sensor.community map (particle data of de SDS011 sensor are shown fine). I suspect an issue with the Sensor.Community API since Opensensemap is working as expected. 

APIs enabled are: Sensor.Community, Madavi.de  (HTTPS), Feinstaub-App, OpenSenseMap.

Maybe someone can point me in the right direction?

Current data:
SDS011 | PM2.5 | 1.3 µg/m³
SDS011 | PM10 | 8.5 µg/m³
 
BMP/E280 | temperature | 8.6 °C
BMP/E280 | air pressure | 1020.34 hPa
BMP/E280 | humidity | 67.4 %
 
WiFi | signal strength | -76 dBm
WiFi | signal quality | 48 %
 

Current device status:
Firmware version | NRZ-2020-129ST:1SDK:2.2.2-dev(38a443e)Core:2.6.2=20602000lwIP:STABLE-2_1_2_RELEASEglue:1.2-16-ge23a07eBearSSL:89454af
-- | --
Free Memory | 26192
Heap Fragmentation | 4 %
Last OTA | 5 min, 41s
NTP Sync | 1
NTP Info | 192.168.1.1 (?) reachable: Yes
Time | Fri Feb 14 12:13:13 2020
Uptime | 5 min, 42s
Reset Reason | Software/System restart
 
SDS011 | 18-11-16(8431)
 
only errors
WiFi | 0
OTA Return | 200
Data Send | 1
Data Send Return | 400
SDS011 | 0
 
Number of measurements | 1
Time spent uploading | 384 ms

ricki-z, 2020-02-14:
Did you change the database entry at my.luftdaten.info?
(and you may have replaced the DHT22 with a BME280. The SDS011 is the PM sensor...)

Spaarne, 2020-02-14:
Thanks! I knew it must have been something simple I forgot about. I changed the sensor config page AND OpenSenseMap but forgot about luftdaten.info. That must be it. I will check it out as soon as possible.

Yes of course it is the DHT22 I replaced, sorry for that mistake.

 Thank you so much.

closed 2020-02-17 by Spaarne