opendata-stuttgart/sensors-software #671: beta: crash nach anhaken von  'OLED SSD1306' oder 'OLED SH1106'

tom-r, 2020-02-16:
Ich versuche ein OLED SH1106 an meinen ESP8266 anzuschließen.
Aber wenn ich in der Konfiguration des Sensors bei 'OLED SH1106' oder 'OLED SSD1306' den Haken setze und auf _Speichern und neu starten_ klicke dann stürzt der Sensor ab mit einem Stack Dump (--> "Monitor" Funktion in vscode, siehe unten).
Ich habe schon mit dem Firmware-Flashtool die komplette Konfiguration gelöscht, bzw. ausschließlich einen BME280 aber noch keinen Partikelsensor und noch gar kein Display angeschlossen.
Dann 'auto update' und 'lade beta'. Ich bin also auf der aktuellen Beta-Version.

Jedesmal fliegt der Sensor beim Speichern des Displays ( _Speichern und neu starten_ ) ab
Es kommt schon gar nicht mehr die Anzeige "Zurück zur Startseite" wie normalerweise und nach dem Reboot ist der Haken beim Display wieder weg...
Haken setzen bei den LCD-Displays funktioniert wie erwartet und wird gespeichert.

Wenn man die "Monitor" Funktion in vscode benutzt sieht man einen Stack-Dump...

Das ist nur bei den Beta-Versionen so.

Auto-update **ohne** beta versionen (also vom master branch) geht, dann läuft das Display wie erwartet, allerdings wird später der SPS30 nicht gefunden (... not found at 0x69), nur der BME280...

Bin ich der Einzige oder haben wir ein Problem in den Beta Versionen mit den beiden Displays ?

Könnte das bitte jemand verifizieren ?

Danke und Gruß,

Thomas

Logausgaben vscode-Monitor:
```
 --- Available ports:
---  1: /dev/ttyACM0         'F5521gw - F5521gw Mobile Broadband Modem'
---  2: /dev/ttyACM1         'F5521gw - F5521gw Mobile Broadband Data Modem'
---  3: /dev/ttyACM2         'F5521gw - F5521gw Mobile Broadband GPS Port'
---  4: /dev/ttyS4           'n/a'
---  5: /dev/ttyUSB0         'USB2.0-Serial'
--- Enter port index or full name: 5
--- Miniterm on /dev/ttyUSB0  9600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ws: config page ...
output debug text to displays...

Exception (29):
epc1=0x4000e1b2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffb90 end: 3fffffc0 offset: 01a0
3ffffd30:  402730c8 3ffeee7c 3ffffd60 40207dce  
3ffffd40:  4027c6c8 3ffffda8 3ffeee7c 4020b9e6  
3ffffd50:  402730c8 3ffeee7c 3ffffda8 4020d7a9  
3ffffd60:  3fff5064 000e000f 00000062 3fff5000  
3ffffd70:  000e000f 80007875 00090003 402730c8  
3ffffd80:  3ffe8644 3ffeee00 00000020 80101453  
3ffffd90:  00000483 3ffffe0c 40201a2c 00000000  
3ffffda0:  00000483 3ffeeeb4 3fff3d04 000003ff  
3ffffdb0:  00000007 3ffffdd0 3fff4e48 00000007  
3ffffdc0:  3ffe8644 3fff2c14 00000000 40222fd4  
3ffffdd0:  3ffffe30 3ffffe30 3ffffe10 40202810  
3ffffde0:  00000003 00000001 3fff2bec 4022d08e  
3ffffdf0:  3ffffe30 3ffeeed0 3fff2bec 401000e1  
3ffffe00:  3fff2bec 3ffeeed0 3fff2bec 40202846  
3ffffe10:  6e6f6300 00676966 80ff0000 3fff0f94  
3ffffe20:  3fff2bec 3ffeeed0 3ffeee7c 4020a67d  
3ffffe30:  6e6f632f 00676966 87000001 3fff3800  
3ffffe40:  0000005f 8000666e 3ffffe80 40234049  
3ffffe50:  00000001 00000001 00000001 3ffeef40  
3ffffe60:  00000003 00000001 07ad3700 00000000  
3ffffe70:  40105c6d 07ad37ce 3ffef8f0 00000001  
3ffffe80:  00000001 3ffeeeb4 3ffeee7c 4020a78b  
3ffffe90:  3ffeebfc 3ffef8f0 00000001 40223ba3  
3ffffea0:  40224a59 00000001 3ffef7e8 40224a4e  
3ffffeb0:  3ffeebfc 3ffeebec 3ffeeb14 4020eeba  
3ffffec0:  000000f0 3fffc6fc 00000000 00000020  
3ffffed0:  30534400 3f003131 80fef7e8 00000030  
3ffffee0:  ffffffff 3fffc6fc 00000001 3ffe86a8  
3ffffef0:  00000000 3fffdad0 3fff3a00 000000ff  
3fffff00:  80000000 3fff3800 00e201ff 80000001  
3fffff10:  3ffef6cc 3ffef6d0 0000002f 40101400  
3fffff20:  00000000 80ffff40 3ffeeb00 402225ac  
3fffff30:  80fef6cc 3ffef600 3fff08a8 80234049  
3fffff40:  3fff2900 0019001f 80efeffe 40201900  
3fffff50:  3ffef6cc 80fef6d0 00000000 402218f4  
3fffff60:  80f42400 906614fc feefef00 feefeffe  
3fffff70:  00000000 3fff08a4 00000020 feef010d  
3fffff80:  00000000 00000000 00000001 40100714  
3fffff90:  3fffdad0 00000000 3ffef7a8 3ffef7e8  
3fffffa0:  3fffdad0 00000000 3ffef7a8 40223cb8  
3fffffb0:  feefeffe feefeffe 3ffe86a8 40100679  
<<<stack<<<
)�
mounting FS...
opened config file...
parsed json...
output debug text to displays...
Connecting to ... 
```

tom-r, 2020-02-24:
Zu dem obigen Crash :
ich habe den obigen stack dump analysiert, mit dem EspStackTrace Decoder :
Problem ist die routine display_debug(...), die erzeugt den Abflug. Und zwar systemweit, sobald sie mit einem der beiden OLED Displays aufgerufen wird.
In dem obigen Beispiel, beim "Speichern und neu starten" ist das in der Routine
webserver_config, die den Restart durchführt (display_debug() am unteren Ende des codes, von mir auskommentiert ). Irgendwas ist da drin was dem ESP8266 nicht schmeckt ...

```
static void webserver_config() {
	if (!webserver_request_auth())
	{ return; }

	debug_outln_info(F("ws: config page ..."));

	server.sendHeader(F("Cache-Control"), F("no-cache, no-store, must-revalidate"));
	server.sendHeader(F("Pragma"), F("no-cache"));
	server.sendHeader(F("Expires"), F("0"));
	// Enable Pagination (Chunked Transfer)
	server.setContentLength(CONTENT_LENGTH_UNKNOWN);

	RESERVE_STRING(page_content, XLARGE_STR);

	start_html_page(page_content, FPSTR(INTL_CONFIGURATION));
	if (wificonfig_loop) {  // scan for wlan ssids
		page_content += FPSTR(WEB_CONFIG_SCRIPT);
	}

	if (server.method() == HTTP_GET) {
		webserver_config_send_body_get(page_content);
	} else {
		webserver_config_send_body_post(page_content);
	}
	end_html_page(page_content);

	if (server.method() == HTTP_POST) {
		//display_debug(F("Writing config"), emptyString);
		if (writeConfig()) {
			//display_debug(F("Writing config"), F("and restarting"));
			sensor_restart();
		}
	}
}
```
Wenn man die beiden Zeilen auskommentiert, so wie oben, werden die Flags an den OLED Displays normal gespeichert. Eigentlich reicht zum speichern schon das Auskommentieren der oberen Zeile, aber das 2. display_debug erzeugt dann nach dem writeConfig() nochmal einen Abflug, nur eben nach dem Speichern ...

Gruß,
Thomas

tom-r, 2020-02-24:
2. Problem :
"Immer wenn eines der beiden OLED Displays aktiviert ist/wird, wird der SPS30 nicht (mehr) gefunden"
Ist ein Treiberproblem im Treiber der OLED Displays ESP8266_SSD1306@4.1.0 :
Und zwar wird beim Init des Displays in der connect() Funktion ein Wire.setClock(700000) abgesetzt. Dieser Befehl sorgt dafür, daß der SPS30 nicht mehr "hochkommt"... warum auch immer...
Die connect-Funktion findet sich in SH1106Wire.h und SSD1306Wire.h des Treibers.
```
    bool connect() {
      Wire.begin(this->_sda, this->_scl);
      // Let's use ~700khz if ESP8266 is in 160Mhz mode
      // this will be limited to ~400khz if the ESP8266 in 80Mhz mode.
      Wire.setClock(700000);
      return true;
    }
```
Lösung:
In den beiden Files die Zeilen **Wire.setClock(700000);** einfach auskommentieren wie oben, dann funktioniert sowohl das Display, als auch der SPS30.
Also so :

```
  bool connect() {
      Wire.begin(this->_sda, this->_scl);
      // Let's use ~700khz if ESP8266 is in 160Mhz mode
      // this will be limited to ~400khz if the ESP8266 in 80Mhz mode.
      //Wire.setClock(700000);
      return true;
    }
```

D.h. aber, daß wir vermutlich den Treiber (zumindest diese beiden Dateien) lokal verwalten müssen ?

Gruß,

Thomas

dirkmueller, 2020-09-15:
This issue has been fixed in -B6 or later.

commit 2ad7ea72 (2020-09-01): Merge pull request #772 from dirkmueller/fix_oled_setting_crash

Fix crash on enabling OLED displays in config (Fixes #671)
commit dd7e0991 (2020-09-01): Fix crash on enabling OLED displays in config (Fixes #671)

When the configuration has just been turned on, the rest of
the code will assuem the display has been initialized. However
initialization only happens on boot, which happens only after
save and restart. So make it possible to differenciate between
intended config and current configuration, and execute display
output only against current configuration to avoid crashes.
closed 2020-09-15 by dirkmueller