opendata-stuttgart/sensors-software #672: Temp Difference dht22 and BME 280

jsturm, 2020-02-17:
Just for fun, I attached 2 different sensors to my AirRohr LoLin ESP NodeMcu V3
Here the results:
DHT22	Temperatur		10.6 °C
DHT22	rel. Luftfeuchte	        57.4 %
BMP/E280 Temperatur		14.4 °C
BMP/E280 rel. Luftfeuchte	38.6 %

The temperatur difference is 3.8°C. Is this normal?
Achim
[image]

Adorfer, 2020-02-17:
you are supplying the BME280 with 5V or 3.3V? 
in other words: This regulator on that board can cause quite an effect.

jsturm, 2020-02-17:
It should be 3,3V

VCC  | -> | Pin 3V3
GND | -> | Pin GND
SCL   | -> | Pin D4 (GPIO2)
SDA  | -> | Pin D3 (GPIO0)

Achim

mdietinger, 2020-02-18:
where are your sensors located?
both of them should be well away from any electronics to ensure they are not headed by the electronics.

jsturm, 2020-02-18:
[image]
They are on the other side of the SDS011 and have no connection to the housing.
Today I repalce both sensors an check again

mdietinger, 2020-02-18:
Very nice build :-)
did you do the measurements as well with open housing?
If there is little to no airflow in the case there is an option that the temparature sensors get headed.

jsturm, 2020-02-18:
No I did't. It seems, the BME280 shows wrong measurements. I checked with a "normal" thermometer.
I will change the BME280 later
My stevrenson housing is on thingiverse -> https://www.thingiverse.com/thing:3181167

Adorfer, 2020-02-18:
the BME280 is heating itself when operated constantly, that's pretty normal. 

The offset should be around 1-2 degrees. That you have 4 degrees is a bit strange. 
Please check for a BME280 from a different source. or look how many mA it draws. 
Potentially the onboard VRE is defective. 
https://github.com/letscontrolit/ESPEasy/issues/164

jsturm, 2020-02-18:
If I look into detail, it looks like I have a 5V sensor (4pins - left) connectet to 3.3 V
[image]  [image]

ropeters68, 2020-02-18:
Nice housing (ABS filament?) ! No static electricity issues?

jsturm, 2020-02-18:
PLA - No static electricity issues! (as far as I know).
In my case ist is not exposed to rain or direct sun

jsturm, 2020-02-23:
I think, some sensors are not very exact. I searched and compared all of my temperature sensors, and here are the results.
I measured at the same place at the same time and under same conditions. 
I waited with the readings for 2h to give the sensor time to acclimatize

Measurements on the balony
The meteorological Institute says: 	temp. 7.1° - humidity 81.3% - air pressure 1010.0 hPa
Yellow ESP with DHT22 says: 		temp. 7.2° - humidity 99.9%
Yellow ESP with BME 280 says: 	temp. 9.6° - humidity 72.3% - air pressure 1003.35 hPa
Blue ESP with DHT 22 says: 		temp. 6.8° - humidity 48.7%
Plain ESP with DHT 22 says:		temp. 6.8° - humidity 82.4%
Plain ESP with DS18S20 says:		temp. 6.0° 	
WEMOS D1 with DHT22 says:		temp. 6,94° - humidity 80.36%
Pimoroni Enviro with BMP 280 says: temp. 9.41° - air pressure 1004.27 hPa
A raspberry with DS18S20 says:	temp. 6.56°
China Sensor with DHT11		temp. 6° - humidity 91%
My Thermometer says:			temp. 9.8°
A Voltrcaft IR-230 thermometer says: temp. 6.8°
Measured the surface of a Coin, next to the sensors

Measurements in the living room:
Yellow ESP with DHT22 says: 		temp. 22.0° - humidity 50.10%
Yellow ESP with BME 280 says: 	temp. 25.9° - humidity 30.8% - air pressure 1004.16 hPa
Blue ESP with DHT 22 says: 		temp. 21.0° - humidity 20.5%
Plain ESP with DHT 22 says:		temp. 20.3° - humidity 42.3%
Plain ESP with DS18S20 says:		temp. 21.5° 	
WEMOS D1 with DHT22 says:		temp. 20.75° - humidity 40.10%
Pimoroni Enviro with BMP 280 says: temp. 26.54° - air pressure 1004.89 hPa
A raspberry with DS18S20 says:	temp. 21.5°
China Sensor with DHT11		temp. 19° - humidity 49%
My Thermometer says:			temp. 20.8°
A Voltrcaft IR-230 thermometer says: temp. 21.5°
Measured the surface of a Coin, next to the sensors

My BME 280 is definitely not OK – I will replace it next days

It is known that the Pimoroni Enviro shows wrong temperature becuse of the layout of that hat.

My test arrangement

[image]

jsturm, 2020-02-23:
I think, some sensors are not very exact. I searched and compared all of my temperature sensors, and here are the results.
I measured at the same place at the same time and under same conditions.
I waited with the readings for 2h to give the sensor time to acclimatize

Measurements on the balony
The meteorological Institute says: temp. 7.1° - humidity 81.3% - air pressure 1010.0 hPa
Yellow ESP with DHT22 says: temp. 7.2° - humidity 99.9%
Yellow ESP with BME 280 says: temp. 9.6° - humidity 72.3% - air pressure 1003.35 hPa
Blue ESP with DHT 22 says: temp. 6.8° - humidity 48.7%
Plain ESP with DHT 22 says: temp. 6.8° - humidity 82.4%
Plain ESP with DS18S20 says: temp. 6.0°
WEMOS D1 with DHT22 says: temp. 6,94° - humidity 80.36%
Pimoroni Enviro with BMP 280 says: temp. 9.41° - air pressure 1004.27 hPa
A raspberry with DS18S20 says: temp. 6.56°
China Sensor with DHT11 temp. 6° - humidity 91%
My Thermometer says: temp. 9.8°
A Voltrcaft IR-230 thermometer says: temp. 6.8°
Measured the surface of a Coin, next to the sensors

Measurements in the living room:
Yellow ESP with DHT22 says: temp. 22.0° - humidity 50.10%
Yellow ESP with BME 280 says: temp. 25.9° - humidity 30.8% - air pressure 1004.16 hPa
Blue ESP with DHT 22 says: temp. 21.0° - humidity 20.5%
Plain ESP with DHT 22 says: temp. 20.3° - humidity 42.3%
Plain ESP with DS18S20 says: temp. 21.5°
WEMOS D1 with DHT22 says: temp. 20.75° - humidity 40.10%
Pimoroni Enviro with BMP 280 says: temp. 26.54° - air pressure 1004.89 hPa
A raspberry with DS18S20 says: temp. 21.5°
China Sensor with DHT11 temp. 19° - humidity 49%
My Thermometer says: temp. 20.8°
A Voltrcaft IR-230 thermometer says: temp. 21.5°
Measured the surface of a Coin, next to the sensors

My BME 280 is definitely not OK – I will replace it next days

It is known that the Pimoroni Enviro shows wrong temperature becuse of the layout of that hat.

My test arrangement
[image]

TempSensoren

closed 2020-02-23 by jsturm

RikDrabs, 2020-02-23:
- Here in Belgium we are using only the standard hardware of Luftdaten (SDS011+DHT22), but in great numbers (>3000).

- The humidity sensor contained in the DHT22 ages very rapidly, and starts aging the first day you use it. It goes from measuring normal values, to a maximum of 100% humidity in the same humidity conditions over a time span of 12 to 24 months in a lot of installations. It should not be used, except when brand new, to measure exact humidity value, nor to compare two measuring stations with each other. The only reason why the sensor is present here is to verify the correctness of the PM10 and PM2,5 measurement. The PM measurement is not correct with humidity values > 70%, which is the reason why those measurements are discarded by statisticians calculating mean values for presentation to politicians. If you want your own sensor to count politically, you better replace the DHT22 as often as needed. If you want to use the humidity sensor data in a "weather station" type use, please use another sensor, or replace frequently.

- The temperature sensor is more accurate, and does not deviate. However, measuring temperature can be very tricky, as any temperature measuring device not only measures contact temperature, but also radiation temperature. If something in the near or far vicinity of the sensor is radiating or absorbing infrared radiation, the measurement can be off in plus or minus by several degrees Celsius, depending on the temperature of the radiating surface. For comparison to be exact you need exact situations: the exact same housing in exact the same place, the exact same view on the space around, even preferably under "thermometer hut" with only one sensor at the time, no MCU in the vicinity, etc, etc. And all of these arguments don't account for the absolute accuracy of temperature measurement by the DHT22, which is +/- 0.2 degrees Celsius, so this alone explains a difference of 0.4 degrees Celsius between sensors.

- For the SDS011 (the PM sensor, which you didn't compare) my experience tells me that this sensor is relatively stable over time, but when time is stretched (sensors installed more than 2 to 4 years ago, in some instances the zero pm reading can be shifted up to values in between 140 to 800 µg/m³. This means that while the real PM10 value is for example 10 µg/m³, the measurement says 140 µg/m³ or 800 µg/m³. These sensors are easily spotted on the map as they show permanently red to violet hexagonals in a see of green ones.