opendata-stuttgart/sensors-software #675: Verbindung zwischen SDS011 und ESP8266 fällt nach einigen Stunden aus

Burny4600, 2020-02-18:
Ich habe den Fehlerfall, dass die Verbindung zwischen zwischen SDS011 und ESP8266 nach einigen Stunden ausfällt.
Wenn ich die Sensoreinheit Spannungslos mache, und wieder mit Spannung versorge, funktioniert die Verbindung zwischen SDS011 und ESP8266 für einige Stunden wieder.
Die i2c Sensoren hingegen fallen nicht aus.
[Gerätestatus.pdf](https://github.com/opendata-stuttgart/sensors-software/files/4219807/Geratestatus.pdf)

TinoM, 2020-03-11:
Ich habe dasselbe Problem:
http://www.madavi.de/sensor/graph.php?sensor=esp8266-100017-sds011

( DHT22: durchgehend : https://www.madavi.de/sensor/graph.php?sensor=esp8266-100017-dht)

Reset über die UI hilft nicht, eine Zeit vom Strom trennen schon. Hab jetzt schon ein anderes Netzteil probiert, aber das ändert auch nichts.

Wenn ich den Sensor ausbaue und per USB verbinde um das Log zu kontrollieren, funktioniert dann immer alles einwandfrei.

sumerland, 2020-05-06:
Hier genau das gleiche. Es fing vor ein paar Wochen an, heute der zweite Ausfall. Neustart des Sensors via GUI ohne Effekt. Trennen von der Stromversorgung für wenige Sekunden hilft.

dirkmueller, 2020-08-31:
Gab es ein Gewitter? Die SDS011 stürzen gerne mal bei Blitze/Gewitter ab und antworten dann leider nicht mehr auf jeglichen seriellen input.

goldfischli, 2020-12-23:
Hab das Problem auch. Kein Gewitter hier.
Auffallend ist ein hoher Fehlerzähler:  SDS011 | 3154
Nach dem Debug-log wird nur noch der BME280 ausgelesen, keine Spur vom SDS011.