opendata-stuttgart/sensors-software #676: No Firmware file found

bastifix, 2020-02-20:
Hallo, ich habe die aktuelle Version heruntergeladen, entpackt, kompiliert und geflasht. Alles ohne Probleme soweit. Gehen tut es aber leider nicht, ich bekomme in der Konsole immer folgende Meldung:

```
No Firmware file found, looking for: /firmware.old
Update Failed.
Please upload the latest firmware as '/firmware.bin' on SPIFSS to recover.
No Firmware file found, looking for: /firmware.bin
Update Failed.
Please upload the latest firmware as '/firmware.bin' on SPIFSS to recover.
```

Habe ich etwas vergessen oder woran könnte das liegen?

ricki-z, 2020-02-20:
Da stellen sich jetzt mehrere Fragen:
- Was genau wurde heruntergeladen und entpackt?
- Womit wurde kompiliert und geflasht?

Ich vermute, daß nur der  "airrohr-update-loader" geladen und geflasht wurde. Das ist aber nicht die Firmware. Der Update Loader wird als Zwischenschritt beim Update der Firmware benötigt und erst von dieser nachgeladen. Die Firmware ist zu finden unter https://github.com/opendata-stuttgart/sensors-software/tree/master/airrohr-firmware

bastifix, 2020-02-20:
Ich habe `opendata-stuttgart/sensors-software` als zip runtergeladen und dann `opendata-stuttgart/sensors-software/airrohr-firmware` entpackt.
Daraufhin die Wifi-Daten angepasst in der `ext_def.h` und mit der Arduino IDE die `airrohr-firmware.ino` geflasht.

ricki-z, 2020-02-20:
Sind auch alle Bibliotheken laut Readme.md installiert?
Und vor allem auch alle Einstellungen (Board Typ, Flash mapping, ...) vorgenommen.

bastifix, 2020-02-20:
Ich hab alles nochmal überprüft, aber nichts gefunden, dennoch eine gute Nachricht: Es geht jetzt wieder ganz normal ;-)
Also vielen Dank für die Hilfe!
Ich kann leider jetzt nicht sagen woran es lag...

closed 2020-02-20 by bastifix