opendata-stuttgart/sensors-software #679: Future sensors. How to follow this issue, and the resulting development for future use?

RikDrabs, 2020-02-26:
The addition of new sensors, like PM, CO, CO2, NOx, and other new proposed sensors, is difficult to follow, as it encompasses different issue numbers, sometimes even for the same sensor.
The future direction of the luftdaten project would be clearer if all this information, and the reaction and development following the proposals, would be united in one single issue.  

What also should be clear is which proposal will be followed in what timeframe, and which proposal will NOT be followed, as to prevent waiting for non-realisable future implementations.
Just to aid in planning campaigns to spread new types of sensors, measuring new things ...

marcauberer, 2020-02-26:
I think an issue is the wrong structure for realizing that. I'd suggest a [project](https://github.com/opendata-stuttgart/sensors-software/projects). There's already exists a project, but it is not really up to date.
Maybe even a project for every release?!

RikDrabs, 2020-02-26:
Indeed a project is a better structure for future sensors. Even a project
for every (future) official release. How to continue from here?

I already came across issue 148, 171, 375, 411, 610, 628, 631, talking
about the actual sensors, and the addition of Rain, PM, CO, CO2, and NOx
sensors.

RikDrabs, 2020-03-12:
OK. The last update in the project section is in 2017. Nothing has been added, or updated since then.  Who is going to do the work to put all info together there, since I don't see how to change things there.