opendata-stuttgart/sensors-software #680: Wrong data SDS011 ?

Rewop-jpc, 2020-02-28:
The values in the graphs are very different from the values I retrieve from my sensor ?
Feinstaubsensor-11453096
https://www.sysanalyser.com/stofmap.html

ricki-z, 2020-03-02:
There are some implementations for home assistant systems that use the same data to generate graphs. And until now we haven't heard about any problems.
So this may a problem on your side. But we can't tell much about this as this is a "black box" for us.

Rewop-jpc, 2020-03-03:
I disconnected it for a while. After starting it up again, the graphs now looks much alike. Before your graph looked very unnaturally. Strait horizontal lines with a few spikes. Now it looks normal!

ricki-z, 2020-03-04:
@Rewop-jpc so the issue is solved? Then please close it.

closed 2020-03-04 by Rewop-jpc