opendata-stuttgart/sensors-software #681: Support for light intensity sensors

atze09, 2020-03-02:
I would be very interested in an Implementation of light insensity. Such as TSL2591 or
BH1750.
Light pollution should also ne covered by this awesome project.
Is it possible to do auch an Integration?

ricki-z, 2020-03-02:
Light pollution is a little bit more than adding a light sensor. We would like to implement this if someone could tell us how to measure this correctly.
Adding the sensor itself to the firmware should be relatively simple.

RikDrabs, 2020-03-03:
Measuring light pollution correctly can be done by measuring only the
reflection of the sky into the sensor, not the direct or indirect local
artificial illumination by street or other lamps.
Only measuring in a vertical axis, with shielding of all the rest, is a
correct measure. Further, since the illumination levels are low to very
low, you need to "zero" reference the sensor, which means that from time to
time you need to take a measure in full dark, to see what the zero level of
the sensor really is. That means the application of a shutter to
temporarily obscure the measurement opening to the sky, and then take the
zero measurement. This can either be done in separate sensor hardware, for
example controlled by a separate ESP8266, controlling the shutter, and
taking the zero and real measurement values, subtracting/comparing them,
and transmitting the values to the luftdaten database directly, or
indirectly through the PM sensors ESP8266.
This could be a nice extension of the Luftdaten project !

atze09, 2020-03-03:
> Light pollution is a little bit more than adding a light sensor. We would like to implement this if someone could tell us how to measure this correctly.
> Adding the sensor itself to the firmware should be relatively simple.

Those are two topics i thought the light intensity would be interesting in my home automation and it would be awesome if you could do an integration of such an sensor just as an addition, same as Temperature and humidity.

The other topic is the above mentioned air pollution, which would be great if you put this your roadmap, but i think this is much more time consuming.

ricki-z, 2020-03-06:
I can't see a mention of "air pollution" in one of the messages above. So what do you mean exactly?

atze09, 2020-03-06:
> I can't see a mention of "air pollution" in one of the messages above. So what do you mean exactly?

Sorry for confusion, i meant light pollution.

bipsendk, 2020-05-26:
For light pollution - get inspired from https://www.instructables.com/id/TESS-W-Night-Sky-Brightness-Photometer/ - one of the challenges will probably be calibration ...

zoomx, 2020-06-26:
In Sky Quality meters projects is often used TSL237 that estimates irradiance instead of lux