opendata-stuttgart/sensors-software #684: 4x20 LCD BACKLIGHT

Beppi4U, 2020-03-06:
Gibt es die Möglichkeit mit einem Taster oder per Software über den WEBServer das Backlight des LCD2004 ein aus zu schalten! Erstens um Strom zu sparen und zweitens das Display zu schonen.
Zusatzfrage: Wird der SPS30 Sensiorion den schon voll unterstüzt - heist werden auch alle Messdaten ausgegeben? Auch per Serial?
Noch eine........: Ist der MHZ19 aktiv messend - soll heißen braucht er einen Gasfluss um messen zu können? Wie wird er verkabelt und in Software includiert?   
vielen Dank und Gruß
Andreas

Falls meine Fragen hier nicht hinpassen oder sollen dann wäre ein kurzer Hinweis nett! ;-)

ricki-z, 2020-03-06:
Das Backlight sollte sich über den Jumper auf dem I2C-Adapter ein- und ausschalten lassen. Dort einen Schalter oder Taster anschließen sollte die gewünschte Funktion bieten.
Der SPS30 wird voll unterstützt bezüglich der Messwerte. D.h. es werden alle Messwerte ausgelesen und übertragen. Leider gibt es wohl ein Problem mit der I2C-Bibliothek (der Takt stimmt nicht bzw. kann nicht umgeschaltet werden), die in der stable Firmware verwendet wird. Daher sollte der SPS30 aktuell mit der Beta betrieben werden.
In unserer Firmware wird der MHZ19 nicht unterstützt, man müsste die entsprechenden Änderungen in eine eigenen Fork der Firmware übernehmen. Ob ein Gasfluss benötigt wird, sollte im Datenblatt beim Hersteller stehen.

Beppi4U, 2020-03-06:
Ich werde wohl für das Display einen Schalter einbauen!
Trotz GOOGLE konnte ich noch nicht in Erfahrung bringen ob der Sensor einen Fluss braucht aber er hat eine Messkammer ähnlich des SDS011 also gehe ich fast davon aus......
Der MHZ19 läuft ja über serial .....nur da hängt ja schon der PM Sensor drann.
Bin gespannnt ob das mal jemand einbauen kann. ;-)
Danke für die Rückmeldung und ein schönes Wochenende
Andi

tom-r, 2020-03-07:
> Zusatzfrage: Wird der SPS30 Sensiorion den schon voll unterstüzt - heist werden auch alle Messdaten ausgegeben? Auch per Serial?

Hi,
der SPS30 wird  prinzipiell unterstützt, aber nicht seriell , nur I2C. Seriell arbeitet der mit  fix 115200 baud, und mit Software-Serial ist das zuviel für den esp8266.
Ich würde Dir den Sensor aber nicht empfehlen, der kann zwar offiziell PM1, PM2.5, PM4 und PM10, in der Realität funktioniert aber nur PM1 und PM2.5.
PM4 und PM10 zeigen bei mit zu 99.8% immer 0.00 an.
Und dabei ist es egal, ob I2C oder seriell angeschlossen, das ist auch mit der Original-Sensirion Auswerte-Software (ControlCenter) so. Ich habe sogar Pullup Wiederstände in die Kabel gelötet, bringt aber auch nix.
Schau mal hier, da haben sich schon andere Mühe gemacht das auszuwerten:
https://github.com/paulvha/sps30 
Lies mal das sps30.odt. Ich kann das aus meiner Erfahrung zu 100% bestätigen.
M.E. ist das ein Firmware-Bug im Sensor.
4x20 LCD Display : in der aktuellen Beta ist die Ausgabe auf dem 4x20 LCD Display für den SPS30 "zerschossen" (4 bzw 5 Messwerte in einer Zeile sind zu lang und alles >20 Zeichen wird vom Displaytreiber umgebrochen, erscheint dann wieder links und überschreibt dir den Anfang ...
Ich hab das zwar bei mir schon gefixt, bin aber noch nicht soweit für einen PR.
Gruß,
Thomas