opendata-stuttgart/sensors-software #688: My sensor is not visible on map or in My sensor list

dejanBpetrovic, 2020-03-12:
After I register my sensor with ID, I cannot see it on the map but sensor exists in CVS file. Why is that.

dejanBpetrovic, 2020-03-12:
my sensor id is 1535541

ricki-z, 2020-03-12:
You sensor is shown in the center of this map:
https://maps.sensor.community/#15/44.3837/20.2675
The ID on the map is 22808 (you should see this ID on my.luftdaten.info if you click on data behind the sensor entry).

dejanBpetrovic, 2020-03-12:
Dear Rajko, something is wrong ,that is not my sensor in Lazarevac. My
sensor is located in Šabac, this is link to my sensor on Opensensormap

https://opensensemap.org/explore/5e538ab834ffa2001b856e9f

Dejan Petrović , IT consultant & developer ,

ricki-z, 2020-03-12:
This is the geo location given at registration for the sensor. You can change it at my.luftdaten.info to the correct location (button "settings" right of the sensor entry)

dejanBpetrovic, 2020-03-13:
That's the problem. I don't see this sensor as my own as a matter of fact I
don't have any sensor on my list as you can see on ss I'm sending to you.

ricki-z, 2020-03-17:
Which screenshot?
The sensor was registered with an email address ending with 'bos.rs'.
Could youo send me the email address of your account to rajko (at) sensor . community?