opendata-stuttgart/sensors-software #692: my data.php not received data from sensor

dejanBpetrovic, 2020-03-16:
I can't receive any data from my sensor in my custom API. 
Here is my sensor configuration and down below is mu data.php code.
What is wrong?

[image]

`<?php

        $retVal = json_decode(file_get_contents('php://input'),true);
        $fp = fopen("C:/xampp/htdocs/MagarenceEvi/data.txt","a+");
        fputs($fp,"-----" . date("d.m.y H:i",time()) . "---\n");
        fputs ($fp,$retVal["esp8266id"] . "\n");
        fputs ($fp,$retVal["software_version"] . "\n");
        foreach ($retVal["sensordatavalues"] as $sensordatavalues) {
                $values[$sensordatavalues["value_type"]] = $sensordatavalues["value"];
        }
        $line = var_export($values,true);
        fputs($fp,$line);
        fclose($fp);

?>`

ricki-z, 2020-03-17:
localhost is a "predefined" IP address (IPv4: 127.0.0.1, IPv6: ::1) and is bound to the local host. So if you use localhost as a hostname in the sensor configuration the sensor will send the data to itself (and not the target host). You need to use the IP address (or the hostname != locahost) of the system running your script.

dejanBpetrovic, 2020-03-17:
Thank you ricki-z , this was a silly thing I did.

closed 2020-03-17 by dejanBpetrovic