opendata-stuttgart/sensors-software #693: Korrekturfaktor für dnms Lärmmesser einstellen

atze09, 2020-03-17:
Hallo,
Kann mir bitte jemand helfen wie ich den Korrekturwert des Lärmsensors bestmöglich einstellen kann. Ohne einen Abgleich mit einem anderen Mikrofon durchzuführen?
Danke.
Freundliche Grüße,
Dominik

Franz4596, 2020-03-18:
hi atze09,  
ich habe Jan'20 ein DNMS (ICS43434 mit Teensy3.6) mit Schallpegelmessgerät verglichen, ab LAeq 42dBA ist keine Korrekrur notwendig (Abw.<1dB). Meines Wissens ist es nicht erforderlich das digitale Mikro zu kalibrieren (Korrekrur=0). Pegel unter 42 bis 37dBA haben deutliche Abweichungen, minimale Anzeige ist ~37dB:
[image]
mfg franz

atze09, 2020-03-19:
Hallo Franz,
Vielen Dank für deine Arbeit. Das ist genau das was ich gesucht habe!
Freundliche Grüße,
Dominik

closed 2020-03-19 by atze09