opendata-stuttgart/sensors-software #694: DHT22 wir nicht eingelesen.

philipplindner-media-network, 2020-03-19:
mein DHT22 mit den Feinstaunsensor SDS011, wird der DHT22 nicht ausgelesen. Volgende Version ist auf den ESP: NRZ-2020-129 im "log" Steht dann folgndes:
`## Sending to sensor.community - SDS011
Succeeded - api.sensor.community
DHT11/DHT22 read failed
DHT11/DHT22 read failed
DHT11/DHT22 read failed
DHT11/DHT22 read failed
DHT11/DHT22 read failed
----
## Sending to madavi.de: 
Succeeded - api-rrd.madavi.de
## Sending to opensensemap: 
Succeeded - ingress.opensensemap.org
## Sending to custom api: 
Succeeded - 192.168.178.23
Time for Sending (ms): 978
ws: data json...
PM10:  23.30
PM2.5: 22.30
----
## Sending to sensor.community - SDS011
Succeeded - api.sensor.community
DHT11/DHT22 read failed
DHT11/DHT22 read failed
DHT11/DHT22 read failed
DHT11/DHT22 read failed
DHT11/DHT22 read failed
----
`
gibt es da eine Fehler in der Softwer oder hat ist mein sensor kaput?

ricki-z, 2020-03-20:
Einen Fehler in der Software würde ich gerade ausschließen. Aktuell sind mit dieser Version über 7000 Geräte mit DHT22 aktiv und senden Daten.
Am besten zunächst die Kabel prüfen. Das Protokoll, das der DHT22 nutzt ist etwas kritisch im Timing.

philipplindner-media-network, 2020-03-20:
danke für deine Antword ich habe eine euen Bestelt und angeschlossen und es gibt wider daten aus.

closed 2020-03-20 by philipplindner-media-network