opendata-stuttgart/sensors-software #697: Sensor zeigt nur mehr Werte zwischen 0.4 und 0.9 ug/m3 ???

cbeham, 2020-03-22:
Hallo,
der Sensor zeigt seit ca. 4 Monaten nur mehr Werte und 1ug/m3.
Habe heute den Sensor zerlegt, da ich mir dachte das vlt. die Ansaugung verstopft ist.
Er gibt aber auch im offenem Zustand nur diese geringen Werte.
Was könnte das sein?

Danke  und lg

[image]

ricki-z, 2020-03-22:
Could you check if the fan is working correctly. And is a tube connected to the sensor? Check also without the tube.

cbeham, 2020-03-22:
hello i tried also with open case. (no tube, only sensor case)
fan is working short time when sensor is starting.
pm 10 and pm 2.5 having the same values.

Franz4596, 2020-04-02:
I had same problen at two devices, after de-Bug-ing (cleaning) the tube, one SDS worked well. You need a small grid in front of the tube to save from bugs. 
Second was clean, Ventilator is running Laser lights, seems do be all ok, but is shows only less values 1-10. I replaced this damage device.