opendata-stuttgart/sensors-software #698: Sensor is removed from sensor community map after maximize intervall

atze09, 2020-03-23:
Hello,
I have a question concerning the sensor community map.
I startet with my noise sensor yesterday with an measurement intervall of 1 minute, i saw that this was too much date, so i switched to half an hours.
Now my sensor is gone from the map.
Is there i maximum intervall in that the sensor appears in the map.
i am currently trying with 15 minutes.

Thank you in advance.
Best regards,
Dominik

ricki-z, 2020-03-24:
We show only sensor on the map that have sent data within the last 5 minutes.
To have comparable measurements the interval should be 145 seconds. This is the default for all devices with our firmware.

Marwe, 2020-03-30:
duplicate of #643 
answered, so closing

closed 2020-03-30 by Marwe