opendata-stuttgart/sensors-software #700: Registration portal crash

RikDrabs, 2020-03-30:
When connecting to meine.luftdaten.info, and trying to show "My sensors" the portal crashes with the message "Internal server error". 
As I told before I'm installing / repairing / maintaining a lot of sensors in the Flemish part of Belgium (LUCHTPIJP), and to be able to follow-up some of the sensors I register them in my name, and do NOT handover them to user's e-mail address. So if there is a problem with them, Rajko's notification mail ("Notification about sensor xxxxx (ChipID: esp8266-xxxxxxxx") is coming in my mailbox, not the mailbox of the user, so I know there is a problem, and can take appropriate actions.
As a result of this, I have a lot of sensors in my sensor list (maybe more than 100 ?).
Until the access problems to the portal started in the mid of the previous week, all was fine. I didn't add a sensor since then (the portal was inaccessible), and when i tried this morning, i saw the portal was on-line again, but with this error message.
By the way, is there any progress in another method for maintenance?
Dual e-mail addresses in the settings would be a big step forward, so that Rajko's mail can be send to the owner of the sensor, AND to the maintenance team. In that case, a handover can be done without losing all knowledge about inoperative sensors.
The second e-mail address could be choosen from a drop-down list, with all e-mail addresses from the supporting groups, and their cities, so users can choose which support group to adhere to.
And thanks for the good work !

ricki-z, 2020-04-01:
We are working on this. As the traffic has increased massively in some data centers the internal networks are sometimes overloaded.
For the moment it seems like we could solve this for the registration page, but not for our main database.

ricki-z, 2020-04-01:
We are planning some changes to our database and API. A second email address for notifications is one of these changes. But not as dropdown list. At the moment there are 31 local groups registered on the map and many are missing there. So the dropdown would be very long.

DeeKey, 2020-07-19:
Solved? Can be closed? @ricki-z @RikDrabs

RikDrabs, 2020-07-19:
Ok, no dropdown list as asked. But i still don't see a second notification email address, which is important for support groups. So on my view this cannot be closed, unless someone explains to me where that second e-mail address is ???