opendata-stuttgart/sensors-software #701: (DFLD) Lärmsensor integrieren

amcsSH, 2020-03-30:
Hallo,

wir haben einen DFLD Lärm-Messgerät im Einsatz und übermitteln dort hin entsprechend die Daten. Gleiche Daten würde ich auch gern an die Sensor.Community weitergeben, doch aktuell finde ich kein Beispiel wie diese Werte aufbereitet und zu übermitteln sind.
In dem aktuellen Format liegen die Werte im Sekunden Takt vor, quasi "kontinuierliche Messung". 

Das Auslesen der Daten funktioniert einwandfrei, die Übermittlung der Daten an die https://api.luftdaten.info/v1/push-sensor-data/ API ist vorbereitet.
Jetzt stellt sich die Frage: Wie sollen/müssen diese aufbereitet werden? 
Könnt ihr mir weiterhelfen?

Besten Dank.

pjgueno, 2020-03-31:
Ich glaube, das ist hier :

`void add_Value2Json(String& res, const __FlashStringHelper* type, const String& value) {
	RESERVE_STRING(s, SMALL_STR);

	s = F("{\"value_type\":\"{t}\",\"value\":\"{v}\"},");
	s.replace("{t}", String(type));
	s.replace("{v}", value);
	res += s;
}`
und da :

`add_Value2Json(s, F("DNMS_noise_LAeq"), F("LAeq: "), last_value_dnms_laeq);
  add_Value2Json(s, F("DNMS_noise_LA_min"), F("LA_MIN: "), last_value_dnms_la_min);
  add_Value2Json(s, F("DNMS_noise_LA_max"), F("LA_MAX: "), last_value_dnms_la_max);`

Haben Sie schon die LAeq, LAmin und LAmax berechnet?

ricki-z, 2020-04-01:
And we are calculating the LA values per measurement interval (default for our noise sensor is 145 seconds). Otherwise we would need to save far too many values.

amcsSH, 2020-04-12:
Thank you very much for this information.
The SourceCode is always a good documentation source ;)
And we calculate the min, max and mean value for a given time-sequence.
The mean value is calculated against DIN45641.