opendata-stuttgart/sensors-software #704: AP wird nicht aufgebaut

ak68-hub, 2020-04-05:
Hallo zusammen,

nach dem korrekten Aufspielen des CH341 Treibers bei einem nodeMCU V3 und der korrekten Installation des Firmware über den Firmware-Flasher wird leider auch nach mehrmaligem Versuch KEIN AP aufgemacht.
Ein Airrohr-xxx WLAN ist auf meinem Smartphone NICHT sichtbar, auch der serielle Monitor ergibt keine Einträge.

PS das Bord ist davor schon mehrerer Tagen mehr schlacht als recht gelaufen mit unzähligen Abstürzen, daher der neue Flash-Versuch, davor hatte ich eine Board Version V2 am Laufen- problemlos.

Versuch mit neuem V3 Board: Hier gelingt die Anmeldung- ist dies ein Hardware-defekt ???

Vielen Dank für Euere Hilfe
Andreas

RikDrabs, 2020-04-05:
Unfortunately not all ESP8266's arriving from China are in 100% working order. I'm doing tech support in Brussels/Belgium for the sensor, and we've shipped more than 3000 kits, but not after flashing and testing each and every one of the ESP8266's: about 5% of them have a problem, from easy things as pins in short-circuit, not being able to be flashed, to the AP not starting, or not connectable. It is very well possible you have a defective chip. Simply use another one, and have more luck. After all, they're cheap.

ricki-z, 2020-04-05:
If there is no output on the serial port then the firmware isn't started. So the flashing doesn't seem to really work.

dirkmueller, 2020-09-02:
please make sure you have your serial port monitor set to the right values:

https://github.com/opendata-stuttgart/sensors-software/blob/master/airrohr-firmware/Readme.md#debug-via-usb-serial

also typically there is a blue led on the board, does it blink or anything?

also, can you maybe post a picture of the exact board you have?