opendata-stuttgart/sensors-software #705: Sensor data

thomas00000000, 2020-04-11:
2 thinks first I can't see my sensor on the map (by the way I normally see a lot of sensors) and second I want to have an indoor display with the data on raspberry pi or esp

ricki-z, 2020-04-12:
There was a database crash last night. The recovery needed longer than expected as the data centers ar a little bit overloaded at the moment. But now it should work again.

RikDrabs, 2020-04-14:
Do you know about:
https://github.com/passantedimezzo/misc/blob/master/AirrohrDisplay.ino
It's a ESP8266 with an LCD 16*2 display. 2 simple building blocks, 4 wires & some firmware.
!! Due to some changes on the server side it's not working anymore. It's also containing a bug concerning the Json decoding.
I have a corrected version in attachment (rename .txt into .ino)
[LCD_airrohr.txt](https://github.com/opendata-stuttgart/sensors-software/files/4473575/LCD_airrohr.txt)
If that project is too simple for you, i have developed a second (much bigger) display with 16 blocks 8*8 dot matrix with a total length of 52 cm, which can be read off a distance of 5 to 20 meters. It is meant for use in public spaces & is a bit more complicated to construct.

thomas00000000, 2020-04-14:
Ok thanks I will have a look at it.

Thomas

dirkmueller, 2020-09-02:
I assume the original issue is resolved (issue with db on sensors.community)

closed 2020-09-02 by dirkmueller