opendata-stuttgart/sensors-software #707: SDS sensor PM data not received when self flashing the NodeMCU

badea-andrei, 2020-04-12:
I've flashed the sensor myself using the latest version of Arduino IDE (1.8.12) with the code on the master branch. When I set it up I've unchecked the autoupdate option and everything was working fine except the values from the sds sensor which did not register in my custom API. So to see if it's a problem with my setup I've checked the auto update option and restarted the sensor and after this the values were sending to the API. I want to do some custom changes on the code and that's why I want to be able to flash the nodeMCU myself. All the above steps were made without any changes to the code. Could this be from my Android IDE setup? I'm suspecting the libraries here.

ricki-z, 2020-04-13:
The version in the master branch is exactly the version used for the actual stable firmware image.
So maybe there was a change in one of the libraries. Please check that exactly the lib versions are used that are mentioned in the platformio.ini.

badea-andrei, 2020-04-13:
Got to the bottom of it.

It was [ESP8266 für Arduino](https://github.com/opendata-stuttgart/sensors-software/blob/master/airrohr-firmware/Contributing.md#build-using-arduino-ide) (Version 2.6.2). I had the newer version installed, 2.6.3.

Thanks for the hint.

closed 2020-04-13 by badea-andrei