opendata-stuttgart/sensors-software #708: GPS NEO-6M - Daten werden nicht aufgenommen

knalla66, 2020-04-22:
Hallo, ich habe in der Configuration gesehen dass NEO-6M unterstützt wird und am Pin D5 und D6 unterstüzt werden. Zustätzlich habe ich ein Display SH1106, SDS001, DHT22 und eben den GPS Sensor NEO-6M im Einsatz. (würde zwar gerne den BME280 statt dem DHT22 einsetzen, jedoch müsste ich dann auf das Display verzichten sofern ich richtig mitbekommen hab)

Es werden alle Daten angezeigt bis auf die GPS Daten.  (über die "value-page" aber auch OLED Display)=) Ich habe schon ein anderes gleiches GPS Bauteil benutzt und die Pins TX und RX getauscht, falls ich mich da versteckt habe - jedoch keine Änderung.

Habe ich einen Denkfehler oder muss ich in der Config händisch noch etwas ändern?

Danke schon mal für eure Hilfe.

ricki-z, 2020-04-29:
Das Display und der BME280 werden beide als I2C Device angeschlossen. I2C ist ein serieller Bus, bei dem sich alle Devices die gleichen Leitungen teilen (außer evtl. die Betriebsspannung).

Der GPS-Support ist absolut EXPERIMENTELL!!! Eigentlich ist die NodeMCU nicht schnell genug, um 2 serielle Geräte zu verarbeiten (also GPS + SDS011 gleichzeitig). Das führt normalerweise zu sehr vielen Resets ud Abstürzen.

Auf dem Display zeigen wir inzwischen schon bis zu 5 "Seiten" an, für GPS würden nochmal mindesten 2 dazu kommen. Da aber das GPS nicht wirklich genutzt werden kann, wurde dafür keine Anzeige umgesetzt. Auf der Seite "Aktuelle Werte" sollten die GPS Infos aber eigentlich angezeigt werden. Voraussetzung ist, das der NEO-6M auch in der lokalen Sensor-Konfiguration aktiviert ist.

knalla66, 2020-04-30:
ok, ich habe jetzt alles abgesteckt bis auf das GPS NEO-6M Modul, damit ich die Last als Fehlerquelle ausschließen kann. Ich habe über die lokale IP auf die Config Page zugegriffen und GPS-NEO 6M aktiviert. reicht diese Aktivierung nicht aus?

[image]

ricki-z, 2020-04-30:
Das sollte reichen. Dann sollten die Werte eigentlich auf der Seite "Aktuelle Werte" angezeigt werden. Zumindest sollten die Zeilen Latitude, Longitute usw. zu sehen sein, wenn auch vielleicht ohne Werte dahinter.

knalla66, 2020-04-30:
ja, aber keine Werte. So war es auch schon als ich die anderen Sensoren angeschlossen hatte.
Aber die Frage war ja warum keine Werte kommen?

ricki-z, 2020-05-01:
Wurden die Neo-6M eventuell vorher mal mit einer anderen Software genutzt? Es gibt die Möglichkeit, das Protokoll von NMEA auf ein uBlox-spezifisches umzuschalten, was die Software eventuell getan haben könnte.
Und es kann bei den Neo-6M teilweise über eine halbe Stunde dauern, bis die Position ausgegeben wird. Gerade wenn die Module längere Zeit nicht in Betrieb waren oder die letzte dem Modul "bekannte" Position etwas weiter weg war. So habe ich das zumindest beim Anschluß an einem Raspberry PI mehrfach erlebt. Ebenfalls wichtig ist die Ausrichtung der GPS-Antenne. Zu Beginn sollte diese möglichst viel Himmel "sehen" können. Am Besten also an einem Fenster testen.

knalla66, 2020-05-05:
nein, sind nagelneu gekauft und das erste Mal gentutzt worden. Ein Modul hatte ich schon bei einer Ausfahrt dabei (mit Mobile Hotspot) für 1h am Rad - bekam da auch keine Werte.

Nordrunner01, 2020-05-05:
Moin.

Wir hatten schon Module dieser Art, wo die serielle Schnittstelle mit 4800 BpS configuriert war, statt mit 9600.

Gruß

ricki-z, 2020-05-07:
Hast du einen USB-2-Serial-Konverter, mit dem man den Neo-6M mal an einen PC anschließen könnte? So einer kommt normalerweise beim SDS011 mit.
Dann sollte z.B. der Serielle Monitor der Arduino IDE zeigen, was der Neo-6M ausgibt.

knalla66, 2020-05-07:
Danke für die Tipps. hab in mal mit dem USB-2-Serial-Konverter am Laptop angesteckt.
jap, da zeigt er Daten an, jedoch kann er keine Position stabilisieren. hab mich am Balkon mit dem Laptop hingesetzt und folgendes im Serial Monitor mitbekommen:
[image]

laut GPGGA keine Long oder Latitude bzw. keine fixe Qualität des Signals. (hab einen 2. Sensor getestet um die Fehlerquelle "Material" zu minimieren)
Ist der GPS NEO-6M so sensibe, hat die Antenne etwas (dann hätten beide Sensoren etwas) oder sind zu wenige Satelitten da (Stadt Wien)? weil wenn man in ruhender Position kein ausreichendes Signal bekommt wie soll man den Sensor in Bewegung einsetzen können?

ricki-z, 2020-05-07:
Falls du ein Android Smartphone hast, kannst du die "aktuelle Lage" mit dieser App prüfen:
https://play.google.com/store/apps/details?id=com.eclipsim.gpsstatus2&hl=de
Die ist relativ einfach gehalten und zeigt die Signalstärke recht übersichtlich an. Damit kannst du acuh testen, wie lang dein Smartphone braucht, um eine Position zu finden. Normal sollten so um die 5 Minuten sein.
Zur Antenne des Neo-6M: Die ist wirklich etwas "sensibel". Die meisten Antennen haben auf dem Keramikkörper einen gut sichtbaren Lötpunkt, der sollte Richtung Himmel zeigen.

knalla66, 2020-05-07:
ok, danke dir. werd ich gleich mal ausprobieren.