opendata-stuttgart/sensors-software #709: Daten an InfluxDB senden

knalla66, 2020-04-22:
Meine InfluxDB ist mit Portforwarding öffentlich erreichbar über https.
Ich verwende die Option "https" in der Configuration (in einem Wiki hab ich gelesen das https nicht unterstützt wird, stimmt das?).
Ich habe in der InfluxDB die verwendete Datenbank erstellt und damit sollte automatisch die measurements anlegen, stimmt das so oder muss ich no etwas einstellen?

Wenn ich den Sensor laufen lasse werden jedoch keine measurements angelegt.

Arbeite das erste Mal damit. Ich nimm Daten mobil über einen Hotspot auf und möchte die Daten gerne direkt in die InfluxDB speichern und später mit Grafana visualisieren.

Wäre für jede Hilfe oder auch Anleitung froh. Danke =)

felixbro, 2020-04-27:
Ich würde mich gerne einreihen!
Genau das gleich Problem habe ich auch!
Bei mir befindet sich der Server hinter einem R-Proxy, der leitet 80 und 433 auch erfolgreich an andere Ports weiter (Grafana läuft ja auf 3000). 
Scheinbar nur 8086 nicht. Wenn ich eine die lokale IP des Servers angebe klappts.

ricki-z, 2020-04-28:
Das erste Problem könnte die Zertifikatsgröße sein. Die NodeMCU kann nur mit 2048-Bit-Zertifikaten arbeiten. Größere sind zu groß für den Speicher der NodeMCU.
Ist in der lokalen Sensor-Konfig auch der richtige Port eingestellt, über den Requests vom Port Forwarding erwartet und dann weitergeleitet werden?
Und soweit ich mich erinnern kann (wir haben den Teil schon lange nicht mehr geändert) ist InfluxDB nicht mit SSL nutzbar, so wie im zweiten Kommentar erwähnt. Sonst gäbe es einen "SSL-Switch" bei der InfluxDB-Konfiuration.

felixbro, 2020-04-28:
Grundgesetzlich habe ich es zum laufen bekommen:
Gebe ich Port 8086 für den influx-server an meinem Router direkt frei, funktioniert alles. 
Meine Domains sind SSL Verschlüsselt, setzte in der MCU den Hacken bei HTTPS funktioniert es nicht.

Offene Ports soll auch der R-Proxy übernehmen. Gebe ich 8086 für den R-Proxy frei funktioniert es nicht.

ricki-z, 2020-04-29:
Wir senden die Query in einem Format, das direkt an die InfluxDB gehen sollte.
Der InfluxDB-Server sollte also so konfiguriert sein, daß er HTTPS Requests annehmen kann. Ein gleichzeitig auf dem Server laufender Webserver spielt dabei keine Rolle.
Soll der Request über den Webserver als Proxy an die InfluxDB geleitet werden, muss als Port dann der des Webservers (also wahrscheinlich 443) angegeben werden und auch der Pfad für den Request muss dann wahrscheinlich geändert werden.
Bei so einer Konfiguration sollte man genau wissen, was wo wie weitergeleitet wird. Gedacht ist die Option eigentlich für eine InfluxDB im gleichen Netz wie der Sensor, wo dann auch nicht unbedingt SSL genutzt wird.

ricki-z, 2020-09-02:
@knalla66 @felixbro Noch ein Hinweis zu HTTPS: Auf Grund des bei der NodeMCU verfügbaren RAMs funktioniert HTTPS nur, wenn der Zielserver (also der, der in der Konfig eingetragen ist) ein max. 2048 Bit langes Zertifikat nutzt.

dirkmueller, 2020-09-02:
es scheint noch ein anderes problem mit https bei influx zu geben. ich kann das problem hier reproduzieren.

knalla66, 2020-09-03:
@ricki-z Es wird genau ein 2048 Bit langes Zertifikat genutzt.

dirkmueller, 2020-09-03:
ja, das sollte gehen. die eigentliche limitierung ist dass wir im moment nur den support für sehr wenige SSL ciphersuites eincompiliert haben. genauer gesagt nur diese:

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.cpp#L827-L830

```
    BR_TLS_RSA_WITH_AES_128_CBC_SHA256,
    BR_TLS_RSA_WITH_AES_256_CBC_SHA256,
    BR_TLS_RSA_WITH_AES_128_CBC_SHA,
    BR_TLS_RSA_WITH_AES_256_CBC_SHA,
    BR_TLS_RSA_WITH_AES_128_CBC_SHA256,
```

dirkmueller, 2020-09-03:
(für details: https://bearssl.org/support.html#guiding-rules )

bb-Ricardo, 2020-12-07:
Dafür habe ich gerade ein issue eingestellt #846. Und erst danach dieses issue gefunden.

Benutze selbst ein LE 4096 Bit Zertifikat und es funktioniert ganz prima.

In neueren nginx Versionen kann man einfach folgende Zeile zur bestehenden Konfiguration hinzufügen:
```
ssl_conf_command Ciphersuites TLS_RSA_WITH_AES_256_CBC_SHA256;
```
Dies betrifft alle SSL/TLS Verbindungen die mit der AirRohr FW aufgebaut werden können.

Hoffe das hilft jemandem weiter.

ricki-z, 2020-12-11:
Die Zertifikate können funktionieren. Nach dem nächsten re-issue kann es damit aber schon wieder vorbei sein. Das Problem hatten wir hier mit unseren Servern, weshalb wir dann auf Zertifikate mit max. 2048 Bit gewechselt haben.

sidietz, 2021-06-10:
May someone link that to the eduroam issue?

It seems to be some general RSA 4096 problem/related issue.