opendata-stuttgart/sensors-software #712: Empty chart in AirTube

IvanKarchev, 2020-04-29:
Hello guys, I have strange problem with data in AirTube website. Sensor working correctly but there is no data in datatable.
[image]
Can you guys help me to solve this problem. I try to reach there support but nobody answered.
You can check my sensor.
[image]

And this is the coordinates
[image]

Petarkir2000, 2020-04-29:
The issue you should address to airtube.info - info@airtube.info

ricki-z, 2020-09-02:
@IvanKarchev can we close this issue? The problem isn't related to a website we are maintaining. So we can't solve this.

dirkmueller, 2020-09-02:
Please report to the airtube.info site maintainers. Thanks for your understanding!

closed 2020-09-02 by dirkmueller

IvanKarchev, 2020-09-03:
Yes, close it.
I need to clear the sensor tube coz it give the high values.
Thank you, best regards,
ivan