opendata-stuttgart/sensors-software #714: PM sensor readings out of whack

BSOD76, 2020-05-02:
Just completed my build. Everything seem to work fine except my PM readings are seem way inaccurate and unrepeatable.  Readings vary by power source, cables used (and cable length). 
Values are high even if I put the sensor assembly inside a CLEAN plastic container. 

If I connect the SDS011 to the windows computer with the usb to serial adapter and measure with DustViewer1.3 [(https://github.com/Galch/DustViewerSharp)](url) the measurements seem reasonable and fairly repeatable , PM 10 = ~8µg/m3, PM 2.5=~7 µg/m3. If I make some dust the values go up as expected. 

Under the same conditions with the nodemcu I get readings PM 10 > 30-600!!µg/m3 and PM 2.5>30-600!!!µg/m3. 
[image]

I've measured the 5V on the sensor an that is not dependant on the usb cables/power source.

BSOD76, 2020-05-02:
Remeasured voltages and measurement data is dependant on it. At this point I'm beginning to think that all the participants measure voltage instead of Particulate Matter.

ricki-z, 2020-05-02:
Where did you measure the voltage?
Some USB power supplies are limiting the current to 100mA (with constant 5V). This may also lead to high values. The laser LED needs a constant current of 100mA, but there is also the NodeMCU that needs some.

BSOD76, 2020-05-03:
Since the "sensitive part"  is the SDS011, I've measured 5V at its inputs.
I had tried two USB phone chargers (2A), a powered USB hub (dedicated 1,5A charging ports), computer USB ports, various cables and the USB to serial adapter.
Only the USB to serial adapter gives sensible readings with DustView.

BSOD76, 2020-05-08:
I have connected a variable output power supply, and set it to get just shy above 5V (5.1) at the PM sensor, compensating the cable length and other losses. The readings are still way over what sensible is, 10-200times of the serial connections data.

ricki-z, 2020-05-08:
I've got your mail. What module type are you using exactly?
Here I could 'see' our defalut firmware version , but in your mail you wrote about an ESP32?

BSOD76, 2020-05-08:
Sorry, I've mixed it up. It's not ESP32, its a LoLin NodeMCU V3 (+SDS011+DHT22, the recommended components).

dirkmueller, 2020-09-02:
the screenshot says you're using a SDS021. which one is true? SDS011 or SDS021?