opendata-stuttgart/sensors-software #715: Split up code to ensure better overview

marcauberer, 2020-05-06:
Is there anything against splitting the source code into several files to keep track of the features?
The `airrohr-firmware.ino` has about 4.5k loc, which is really large in my opinion.

ricki-z, 2020-05-07:
This may be done step by step. Its sometimes difficult to find all places that need to be changed (i.e. if we add a new sensor) if there are too many files.

dirkmueller, 2020-09-02:
no objection to modularization (I started this a few months ago). I just don't want to modularize with bloating the binary code too much, and most attempts I tried (like moving globals into structures) increase code size quite a bit. 

so we' can only do that after upgrading sketch max size to 1MB. currently we're in a difficult situation where we need to take care of binary size over anything else.