opendata-stuttgart/sensors-software #716: Send to own api does not work anymore

andreas83, 2020-05-07:
Hello, 

i just noticed that my external api will not be  triggered anymore. 

Firmware
```
NRZ-2020-130-B5
ST:1
SDK:2.2.2-dev(38a443e)
Core:2.6.2=20602000
lwIP:IPv6+STABLE-2_1_2_RELEASE
glue:1.2-16-ge23a07e
BearSSL:89454af
```

svenmarcelmueller, 2020-05-17:
I have the same issue. Does anyone know how to fix this?

BR, Sven

andreas83, 2020-05-18:
Its very related to ssl, because it works without.

ricki-z, 2020-05-18:
Sorry for the late answer:
SSL should support TLS 1.2, ciphers should support AES256-SHA256, AES128-SHA256, AES256-SHA or AES128-SHA.
The used certificate should be 2048 Bit maximum size. Greater needs too much RAM.
All this are limitations of the SSL library and the available RAM and cpu speed of the ESP8266.
(other ciphers need too many calculations and may cause http timeouts)

andreas83, 2020-05-19:
@ricki-z thx for clarification.

closed 2020-05-19 by andreas83

svenmarcelmueller, 2020-05-19:
Sorry for bringing this up again, but also without ssl (http) it does not work. 
In my Server Logs I cannot see any traffic from the sensor (I think that there is done a http/POST request to the server with the JSON data in its body) and when I connect via USB to the ESP Board and switch debug info to the max I just see the message that data was sent to the external resource, but not where to and if it was ok or not.
Anybody with the same problem?
Thanks, Sven