opendata-stuttgart/sensors-software #717: Failure to install OTA update

Bergerie, 2020-05-07:
I fail to install install the program via de Arduino IDE using the following procedure:
* create data subfolder in sketch folder
* download latest firmware from https://firmware.sensor.community/airrohr/update/ and put this as firmware.bin in /data
* Put ESP in flash mode through Flash(Hold)-Reset (Hold-Release)- Flash(Release)
* Launch via Arduino IDE within sketch airrohr-firmware.ino, Sketch Dat Upload from Tools menu.
* Upload sketch airrohr-firmware.ino ( https://github.com/opendata-stuttgart/sensors-software/tree/master/airrohr-firmware) 
* Enable in http://192.168.4.1/config#wlanpwd Auto Update Firmware
 
-> twoStageOTAUpdate
-> Update md5: 598a4c371edcc0dc83c92b03d9ebff8b
-> Sketch md5: e9d8b99c8eba47da23f817b2263238af
-> Success downloading: /airrohr/update/latest_en.bin
-> Success downloading: /airrohr/update/latest_en.bin.md5
-> Success downloading: /airrohr/loader-002.bin
-> launching 2nd stage
-> Erasing SDK config.
-> Restart.
-> ?)⸮1⸮⸮⸮⸮⸮No Firmware file found, looking for: /firmware.bin
-> Update Failed.
-> Please upload the latest firmware as '/firmware.bin' on SPIFSS to recover.
-> No Firmware file found, looking for: /firmware.old
-> Update Failed.

Any guidance how to resolve the issue would be appreciated.
Rgds

ricki-z, 2020-05-07:
OTA update = Over-The-Air update
It should be enough to activate auto updates to get the latest stable release.
Otherwise use the flashing tool mentioned on our home page.
If you want to flash the latest image you need to flash it to the beginning of the flash memory and not to the SPIFFS file system (what you have done).
What did you try to do?

Bergerie, 2020-05-07:
Thanks for attending to my issue.

Indeed I do understand what OTA means and as such I prefer to have that set on for the purpose of future releases. 

However, instead of using the flashing tool mentioned on your homepage, I rather wish to opt for uploading the sketch using the Arduino IDE.

In fact, I’m following the VARIANT 1 Arduino IDE as described in https://github.com/opendata-stuttgart/meta/wiki/EN-Installing-firmware.

I followed the instructions for connecting to my wifi network, using the default values in the config file, including the the enabled option OTA firmware update. The following is the result, leaving me with the error message Update failed.

 

21:18:45.690 -> twoStageOTAUpdate

21:18:47.805 -> Update md5: 598a4c371edcc0dc83c92b03d9ebff8b

21:18:47.840 -> Sketch md5: 69077cca9e7ced90e4a37f38ae1efdac

21:19:10.468 -> Success downloading: /airrohr/update/latest_en.bin

21:19:11.512 -> Success downloading: /airrohr/update/latest_en.bin.md5

21:19:24.576 -> Success downloading: /airrohr/loader-002.bin

21:19:25.812 -> launching 2nd stage

21:19:29.413 -> Erasing SDK config.

21:19:29.646 -> Restart.

21:19:30.252 -> ⸮⸮⸮⸮1⸮⸮⸮⸮⸮No Firmware file found, looking for: /firmware.bin

21:20:02.720 -> Update Failed.

21:20:02.753 -> Please upload the latest firmware as '/firmware.bin' on SPIFSS to recover.

21:20:11.081 -> No Firmware file found, looking for: /firmware.old

21:20:11.115 -> Update Failed.

21:20:11.148 -> Please upload the latest firmware as '/firmware.bin' on SPIFSS to recover.

 

Could you please provide me with instructions how to overcome this.

Regards

Dick Wismeijer

ricki-z, 2020-05-07:
If you want to do this that way, then you need to check that all parameters like NodeMCU type, SPIFFS layout and size, cpu task and so on are correctly configured (mentioned in Readme.md).
And then you need to click on "Sketch -> Upload", but you wrote in your issue that you clicked on "Sketch Data Upload" which is found under "Tools" and not under "Sketch".

Bergerie, 2020-05-07:
"Sketch Data Upload" which is found under "Tools" and not under "Sketch" - that's what I did in order to load a data file initially;
"And then you need to click on "Sketch -> Upload", "- that's what I did after your 1st response.
In both cases I get the response: "Please upload the latest firmware as '/firmware.bin' on SPIFSS to recover."
"all parameters like NodeMCU type, SPIFFS layout and size, cpu task and so on are correctly configured (mentioned in Readme.md)" I have been looking for them but I can't find them in any of the read.me files contained in the sensors-software-master portfolio. Any change you summarize separately for me?

ricki-z, 2020-05-07:
Most important:
Board: NodeMCU 1.0 (ESP12-E module)
CPU frequency: 160MHz
Flash size: 4MB / 3MB SPIFFS(filesystem)

I think the last one isn't configured correctly and the filesystem is too small for the downloaded firmware image (+  loader + backup of old fireware).

Bergerie, 2020-05-08:
Thanks a lot, worked perfectly well. The flash size was the issue. Thank you so much. Issue closed.

closed 2020-05-09 by ricki-z